pycom/pymakr-atom #1: Failed to load the Pymakr package

robert-hh, 2017-04-13:
Installed, started, and the: red error Box:

Failed to load the Pymakr package
Cannot find module 'serialport'

System debian Wheezy
Atom 1.16.0

### Stack Trace

Failed to load the Pymakr package

```
At Cannot find module 'serialport'

Error: Cannot find module 'serialport'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (/usr/share/atom/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._resolveFilename (/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/connections/pyserial.js:4:16)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/board/pyboard.js:3:34)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/main.js:7:21)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
```

### Commands

```
```

### Non-Core Packages

```
Pymakr 0.9.2 
```

robert-hh, 2017-04-13:
Deleting both Atom and Pymakr and reinstalling both solved the issue.

franckalbinet, 2017-04-13:
Got the same error, tried to re-install both Atom and Pymaker plugin but still the same error.

**Atom**: 1.16.0 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.9.5
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 0.9.2

### Stack Trace

Failed to load the Pymakr package

```
At Cannot find module 'serialport'

Error: Cannot find module 'serialport'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (/Applications/Atom.app/Contents/Resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._resolveFilename (/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/connections/pyserial.js:4:16)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/board/pyboard.js:3:34)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/main.js:7:21)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/app.asar/src/package.js:796:27)
    at /app.asar/src/package.js:123:28
    at Package.module.exports.Package.measure (/app.asar/src/package.js:96:15)
    at Package.module.exports.Package.load (/app.asar/src/package.js:110:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/app.asar/src/package-manager.js:468:14)
    at PackageManager.module.exports.PackageManager.activatePackage (/app.asar/src/package-manager.js:548:30)
    at /app.asar/node_modules/settings-view/lib/package-manager.js:484:29
    at exit (/app.asar/node_modules/settings-view/lib/package-manager.js:99:16)
    at triggerExitCallback (/app.asar/src/buffered-process.js:322:11)
    at /app.asar/src/buffered-process.js:343:11
    at /app.asar/src/buffered-process.js:204:9)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:493:12)
```

### Commands

```
     -4:13.5.0 core:confirm (input.hidden-input)
     -2:36.8.0 core:copy (atom-notification.fatal.icon.icon-bug.native-key-bindings.has-detail.has-close.has-stack)
```

### Non-Core Packages

```
Pymakr 0.9.2 
```

robert-hh, 2017-04-13:
So what I did, after reinstalling Atom to 1.16, I de-installed Pymakr from the newly installed Atom and then installed it again, with my LoPy attached. I must admit that this was a kind of blind shot, before trying to find the real reason and hunting for a module called serialport.

franckalbinet, 2017-04-14:
Thanks a lot for your help! I tried your step but do not work on my side unfortunately. Will re-open an issue.

robert-hh, 2017-04-14:
I just tried it on OS X, 10.11.6. I had same problem. But:
After Atom was started and told me about the missing package, I de-installed Pymakr using Atoms Setting menu and re-installed it again, keeping Atom open. Then the Pymakr worked.

RalphHogenbirk, 2017-04-14:
For debugging this issue, please try to run npm install or apm install from the plugin folder (/users/**/.atom/packages/pymakr/). Remove the node_modules folder before running it. 
I'm curious what kind of error you'll get when doing that. 

If that doesn't throw any errors but still pymakr won't work, try to run the install.js in the root folder (node install.js) and see if that either helps or throws an error that we can use to debug.

franckalbinet, 2017-04-14:
Running `npm install` from `pymakr/` folder after removing `node_modules`,  I get the following error in the console (looks like related to `electron-build`):

```
npm WARN engine electron-rebuild@1.5.7: wanted: {"node":">=6.0.0"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine ora@0.3.0: wanted: {"node":">=4"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine cli-spinners@0.2.0: wanted: {"node":">=4"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine rxjs@5.3.0: wanted: {"npm":">=2.0.0"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine request@2.81.0: wanted: {"node":">= 4"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine har-validator@4.2.1: wanted: {"node":">=4"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine form-data@2.1.4: wanted: {"node":">= 0.12"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine hawk@3.1.3: wanted: {"node":">=0.10.32"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine har-schema@1.0.5: wanted: {"node":">=4"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine boom@2.10.1: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine cryptiles@2.0.5: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine hoek@2.16.3: wanted: {"node":">=0.10.40"} (current: {"node":"0.10.26","npm":"1.4.27"})
npm WARN engine co@4.6.0: wanted: {"iojs":">= 1.0.0","node":">= 0.12.0"} (current: {"node":"0.10.26","npm":"1.4.27"})
\
> Pymakr@0.9.2 install /Users/ganesh/.atom/packages/Pymakr
> node install.js

{ [Error: Command failed: 
/Users/ganesh/.atom/packages/Pymakr/node_modules/electron-rebuild/node_modules/ora/index.js:2
const chalk = require('chalk');
^^^^^
SyntaxError: Use of const in strict mode.
    at Module._compile (module.js:439:25)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/Users/ganesh/.atom/packages/Pymakr/node_modules/electron-rebuild/lib/cli.js:20:12)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:902:3
] killed: false, code: 8, signal: null }
bindings@1.2.1 node_modules/bindings

nan@2.6.2 node_modules/nan

commander@2.9.0 node_modules/commander
└── graceful-readlink@1.0.1

debug@2.6.3 node_modules/debug
└── ms@0.7.2

lie@3.1.1 node_modules/lie
└── immediate@3.0.6

object.assign@4.0.4 node_modules/object.assign
├── function-bind@1.1.0
├── object-keys@1.0.11
└── define-properties@1.1.2 (foreach@2.0.5)

telnet-client@0.13.3 node_modules/telnet-client
└── bluebird@3.4.7

xterm@2.5.0 node_modules/xterm

electron-rebuild@1.5.7 node_modules/electron-rebuild
├── node-abi@1.3.3
├── colors@1.1.2
├── ora@0.3.0 (log-symbols@1.0.2, cli-spinners@0.2.0, cli-cursor@1.0.2, chalk@1.1.3)
├── fs-promise@1.0.0 (thenify-all@1.6.0, any-promise@1.3.0, mz@2.6.0, fs-extra@1.0.0)
├── yargs@3.32.0 (decamelize@1.2.0, camelcase@2.1.1, y18n@3.2.1, window-size@0.1.4, os-locale@1.4.0, string-width@1.0.2, cliui@3.2.0)
├── bluebird@3.5.0
├── rimraf@2.6.1 (glob@7.1.1)
├── babel-register@6.24.1 (home-or-tmp@2.0.0, mkdirp@0.5.1, source-map-support@0.4.14, babel-runtime@6.23.0, babel-core@6.24.1, lodash@4.17.4, core-js@2.4.1)
├── node-gyp@3.6.0 (graceful-fs@4.1.11, semver@5.3.0, osenv@0.1.4, minimatch@3.0.3, nopt@3.0.6, which@1.2.14, fstream@1.0.11, mkdirp@0.5.1, glob@7.1.1, tar@2.2.1, npmlog@4.0.2, request@2.81.0)
├── spawn-rx@2.0.10 (lodash.assign@4.2.0, rxjs@5.3.0)
└── babel-runtime@6.18.0 (regenerator-runtime@0.9.6, core-js@2.4.1)
```

RalphHogenbirk, 2017-04-14:
@franckalbinet The "Use of const in strict mode." is due to your node version being outdated (0.10.26). Running 'apm install' will likely the problem, since atom uses it's own node version. Or you can update your local node version to at least 4.x.

franckalbinet, 2017-04-14:
So after, package install fail `Cannot find module 'serialport'`

1. I upgraded  node to `v7.9.0`
2. remove `/node_modules`
3. `npm install`
4. build is done properly now
5. and package is installed properly (no error during Atom startup) and managed to connect to Lopy module through both Wifi and serial port.

Thanks a lot for your quick support, really appreciated!

However, when uninstalling Pymakr from Atom itself and installing again through Atom UI, I still get the same error type (see below). This is not critical to me as I can use the workaround described above now but would be good to do it using the default way as well.

**Atom**: 1.16.0 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.9.5
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 0.9.2

### Stack Trace

Failed to activate the Pymakr package

```
At Cannot find module 'serialport'

Error: Cannot find module 'serialport'
    at Module._resolveFilename (module.js:455:15)
    at Module._resolveFilename (/Applications/Atom.app/Contents/Resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.Module._resolveFilename (/app.asar/src/module-cache.js:383:52)
    at Function.Module._load (module.js:403:25)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/connections/pyserial.js:4:16)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/board/pyboard.js:3:34)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/main.js:7:21)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
```

RalphHogenbirk, 2017-04-14:
@franckalbinet Good to see you got it working! We still have to figure out what the root cause is of the error when installing through atom.

Since it happens on your machine, can you try again to remove /node_modules and then run 'apm install' instead of npm? This should be exactly the same as installing it through the atom ui, with the difference that you can see the output of any errors that are thrown. Can you try this and tell me if you see anything other than a success message?

franckalbinet, 2017-04-14:
Ran `apm install` after deleting `/node_modules`:

No error, just getting two words in my console: `Installing modules ✓`

But then, while starting up Atom still the same `serialport` error.

FYI, `apm --version` outputs: 
```
apm  1.16.1
npm  3.10.5
node 6.9.5 x64
python 3.6.0
git 1.8.5.2
```

RalphHogenbirk, 2017-04-14:
Thanks for sharing the apm version @franckalbinet... I think i found it because of that. The serialport lib uses node-gyp to compile, which requires python 2.x while you are using 3.6.
You can tell apm to use python 2 with the following line (assuming you have python 2 installed):

echo "python=/usr/bin/python2" >> ~/.atom/.apmrc

Let me know if that indeed solves it. If it does, I'll add it to the readme file and will think about how to automate this without permanently setting the apm python version to 2

franckalbinet, 2017-04-14:
Et voilà ! As you suggested I ran `echo "python=/usr/bin/python2.7" >> ~/.atom/.apmrc` (just looked up to a proper Python 2 version name) and now it works.  Thanks a lot!

closed 2017-04-15 by RalphHogenbirk

paolodemo, 2017-04-18:
I had the same problem and the proposed solutions doesn't fixed it.
I solved installing g++ which seems to be a dependency of serialport:
`sudo apt install g++`

I hope this can be useful.

RalphHogenbirk, 2017-04-18:
@paolodemo On which specific OS and version did this occur? Perhaps I can build in a check and automate this

paolodemo, 2017-04-18:
Linux Mint 18.1 Serena Cinnamon 64-bit
@RalphHogenbirk if you need more specific information, send me the command line.

adambrun, 2017-04-18:
Done all of above - nothing seems to work. I get from apm install a small positive green check mark. But when starting atom after it issues a new error. First when I installed Xcode (many GB) and redone the apm install - it works.

RalphHogenbirk, 2017-04-18:
@adambrun You are right, the serialport library needs xcode to work. Actually it only needs the xCode Command Line Tools. Of you install only those, you should be fine as well (it's 130mb). 

I'm considering trying to pre-compile the serialport library for mac and linux as well, like we did for windows. That will prevent all of the above problems, as long as that one compiled version works on all OS's

adambrun, 2017-04-18:
Let me know if you want me to try it out here.

Your quick response is very appreciated. First the SiPy I ordered was reported to have a deep sleep failure - very informative mail app. 2h after I ordered - then all the trouble to get Atom pymakr to work - all of this is gone when it finally works.

I co-own two startup companies that are currently looking for the right technological platform. we-teco.com and IotKompagniet.dk.

Going for prototyping 100 pcs with SigFox - to see if we should use SiPy as our basis module. Need however to see deep sleep work later in may. 

Adam Brun

+45 2326 0410
adam@adambrun.dk
Adam Brun Projekt IVS CVR: 37173606

Eric24, 2017-04-22:
I'm getting this error on Windows 10. I have only Python 2.7 installed on this machine. The specific "red box" error I'm getting is "%1 is not a valid Win32 application". Anyone else having this particular problem?

danicampora, 2017-04-22:
@RalphHogenbirk I think @Eric24 error reason is because the pre-compiled script for Windows is only valid for the 64-bit version. If that's the case then we need to include the 32-but version as well, and detect the OS type before deciding which one to load. Please have a look at that.

Sygyzmundovych, 2017-04-22:
Just got my LoPy+Board today, upgraded firmaware and tried to install Pymakr Atom Package now. Using Windows 10. Got "%1 is not a valid Win32 application." and error stack:

**Atom**: 1.16.0 ia32
**Electron**: 1.3.13
**OS**: Microsoft Windows 10 Enterprise
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 0.9.3

### Stack Trace

Failed to load the Pymakr package

```
At %1 is not a valid Win32 application.
\\?\C:\Users\I076835\.atom\packages\Pymakr\lib\connections\serialport-win\build\Release\serialport.node

Error: %1 is not a valid Win32 application.
\\?\C:\Users\I076835\.atom\packages\Pymakr\lib\connections\serialport-win\build\Release\serialport.node
    at Error (native)
    at process.module.(anonymous function) (ELECTRON_ASAR.js:168:20)
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:168:20)
    at Object.Module._extensions..node (module.js:583:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:168:20)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at bindings (/packages/Pymakr/node_modules/bindings/bindings.js:76:44)
    at /packages/Pymakr/lib/connections/serialport-win/lib/bindings.js:3:35)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/connections/serialport-win/lib/serialport.js:12:25)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/connections/pyserial.js:8:18)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/board/pyboard.js:3:34)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at /packages/Pymakr/lib/main.js:7:21)
    at Module._compile (/app.asar/src/native-compile-cache.js:109:30)
    at /app.asar/src/compile-cache.js:216:21)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (module.js:483:17)
    at require (/app.asar/src/native-compile-cache.js:50:27)
    at Package.module.exports.Package.requireMainModule (/app.asar/src/package.js:796:27)
    at /app.asar/src/package.js:123:28
    at Package.module.exports.Package.measure (/app.asar/src/package.js:96:15)
    at Package.module.exports.Package.load (/app.asar/src/package.js:110:12)
    at PackageManager.module.exports.PackageManager.loadPackage (/app.asar/src/package-manager.js:468:14)
    at PackageManager.module.exports.PackageManager.activatePackage (/app.asar/src/package-manager.js:548:30)
    at /app.asar/node_modules/settings-view/lib/package-manager.js:484:29
    at exit (/app.asar/node_modules/settings-view/lib/package-manager.js:99:16)
    at triggerExitCallback (/app.asar/src/buffered-process.js:322:11)
    at /app.asar/src/buffered-process.js:352:11)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

### Commands

```
     -9:21.4.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
markdown-preview-plus 2.4.9 
Pymakr 0.9.3 
sap-markdown-preview 0.1.0 
```

So, seems I have 32-bit Atom, but would need 64-bit version to work with Pymark package?

Eric24, 2017-04-22:
@danicampora Hmmm. I'm running 64-bit Windows 10, but atom.exe itself is indeed 32-bit (with no new news on a 64-bit version). What would it take to compile a 32-bit version?

danicampora, 2017-04-22:
@Sygyzmundovych let us try to fix this issue on Monday (we need to make it 32-but compatible as well). Apologies for the inconveniences.

danicampora, 2017-04-22:
@Eric24 OK then the issue is a bit more complex. The plug-in does run on Windows (and on Windows 10 for sure).

Eric24, 2017-04-22:
@danicampora I think I'm having the same issue as @Sygyzmundovych just reported. If it's not a 32/64-bit issue, how to diagnose? It seems strange that we are the only two people trying to run Atom on Windows 10. I suppose I could move to a Mac in the mean time, but it's not my preference.

RalphHogenbirk, 2017-04-23:
@Eric24 The precompiled serialport lib should be suited for 32 and 64 already so that is probably not the issue. 

You were one of the people who had a pre-release version of the plugin, correct? If so, make sure that old version is unlinked and then try again. Do so by either running 'apm unlink' from the pre-release pymakr folder, or remove the symlink to it in the ~/.atom/packages folder. 

Let me know if that works.

Eric24, 2017-04-23:
@RalphHogenbirk I don't think I had a pre-release version of anything. I was using the pymakr IDE until yesterday, when I uninstalled it and I did a fresh install of the Atom package (first time).

If you still think I should try to unlink it, what folder would I be looking for (as I don't see a pre-replease pymakr folder).

RalphHogenbirk, 2017-04-25:
@Eric24 Ok, that must have been another Eric then :)

Strange, because I had the same error and this was caused by having both the official plugin installed and an older version I added to pymakr with 'apm link' (though I'm not quite sure how). Anyway, to make sure, double check the ~/.atom/packages folder if there are multiple Pymakr folders or symlinks. 

Otherwise I'm not sure of a solution for this specific problem. I'll test it on more machines I can get my hand on.

Eric24, 2017-04-26:
Yes, I've double-checked and there is currently nothing in my packages folder.

bobpotter3999, 2017-04-28:
@eric24 Having issues with Windows 10 Version 1607 too. You're not alone :) Tried doing the re-install suggestions but that didn't solve it. May try with Linux Mint or just wait for a WIN 32 version

bobpotter3999, 2017-04-28:
@paolodemo Installed g++ as you suggested on Linux Mint 17.3 and that got round the serial port issue. Nice one . Thanks.

saurubh, 2017-05-03:
@Pycom guys, do you guys really have anything working?i am sick using Lopy with every step getting errors, either related IDEs, plugin, or APIs. Please publish after you are done with your developments.

RalphHogenbirk, 2017-05-03:
Sorry about that @saurubh, but we can only test on so many os versions. After that, we can only try to fix it as problems come in. 

Anyone still struggeling with a missing serialport library, in this weeks update of v0.9.4 we are shipping the precompiled serialport lib for both mac and linux inside the repo. If for any reason it can't find the serialport lib, it switches to the precompiled version. This acts as a backup if the serialport lib is not installed properly for any reason and should solve at least a large part of the installation problems we've seen so far.

Please let me know if anyone still has problems with the library after this update.

kwmartin, 2017-06-05:
Haved tried all of the above suggestions but can not get pymakr plugin working in Ubuntu 16.04. After rm -rf and >npm install in ~/.atom/packages/Pymakr I get module version mismatch, expected 43, got 49.
>node --version and apm --version don't match:
> node --version
v7.10.0
> apm --version
apm  1.0.1
npm  2.5.1
node 0.10.35
python 2.7.12
git 2.7.4
> 
I can access using miniterm.py over /dev/ttyUSB0 but can not access 192.168.4.1 which might be a routing problem. I am connected to WiFi access point as far as I can tell.
> sudo wpa_cli status
Selected interface 'wlan1'
bssid=24:0a:c4:02:76:15
freq=2437
ssid=lopy-wlan-7614
id=1
mode=station
pairwise_cipher=CCMP
group_cipher=TKIP
key_mgmt=WPA2-PSK
wpa_state=COMPLETED
ip_address=192.168.1.210
p2p_device_address=10:fe:ed:12:2d:8b
address=10:fe:ed:12:2d:8b
uuid=54f2867c-bfb9-5a00-9b66-1aefe77b22a2
> 
Can't ftp to 192.168.4.1:
> ftp 192.168.4.1
ftp: connect: Connection timed out
ftp> 
I separately install pymakr outside of atom; can not connect to 192.168.4.1 but can connect to /dev/ttyUSB0
trying with telnet:
> telnet 192.168.4.1
Trying 192.168.4.1...
telnet: Unable to connect to remote host: No route to host
> 
My routing is:
> route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG    0      0        0 eth0
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eth0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.0.0     0.0.0.0         255.255.0.0     U     0      0        0 wlan1
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 eth0
192.168.4.1     0.0.0.0         255.255.255.255 UH    0      0        0 wlan1
192.168.122.0   0.0.0.0         255.255.255.0   U     0      0        0 virbr0
> 
Any suggestions?

RalphHogenbirk, 2017-06-07:
@kwmartin Atom indeed runs a different version of node in most cases. Try to use 'apm install' instead of 'npm install', that might fix it.

I'm not sure about your problems connecting over telnet to the board. Possibly you'll have to remove the 192.168.4.1 line from your routing table. But that's a bit off topic for now, since it's not pymakr related.

Nick-Aus, 2017-06-21:
Clean install (never installed Atom or Py-anything before).
Win 7 Pro, 32-bit.
Installed Atom 1.18.0 from atom.io
Installed Pymakr package 1.0.1 from Atom's "Install a Package" interface.
Pymakr would not load - compiled for earlier version etc. Rebuild, ok.

**Failed to load the Pymakr package
%1 is not a valid Win32 application.**
\\?\C:\Users\Nick\.atom\packages\Pymakr\lib\connections\serialport-win\build\Release\serialport.node

Chris-Nbg, 2017-07-02:
Clean install on Win7 (32-bit), exactly same experience as @Nick-Aus.

RalphHogenbirk, 2017-07-03:
@Chris-Nbg You can try the current code on the develop branch (copy the code to the package folder and make sure to run 'apm install' from that folder in the terminal). Let me know if that works for you.

Chris-Nbg, 2017-07-03:
@RalphHogenbirk: apm stopped with errors while "installing modules ..."

excerpt from the error log:

`[...]
27218 error Windows_NT 6.1.7601
27219 error argv "C:\\Users\\Chris-Nbg\\AppData\\Local\\atom\\app-1.18.0\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\Chris-Nbg\\AppData\\Local\\atom\\app-1.18.0\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\Chris-Nbg\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\Chris-Nbg\\.atom\\.apmrc" "install" "--runtime=electron" "--target=1.3.15" "--arch=ia32"
27220 error node v6.9.5
27221 error npm  v3.10.10
27222 error code ELIFECYCLE
27223 error serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
27223 error Exit status 1
27224 error Failed at the serialport@4.0.7 install script 'node-pre-gyp install --fallback-to-build'.
27224 error Make sure you have the latest version of node.js and npm installed.
27224 error If you do, this is most likely a problem with the serialport package,
27224 error not with npm itself.
27224 error Tell the author that this fails on your system:
27224 error     node-pre-gyp install --fallback-to-build
27224 error You can get information on how to open an issue for this project with:
27224 error     npm bugs serialport
27224 error Or if that isn't available, you can get their info via:
27224 error     npm owner ls serialport
27224 error There is likely additional logging output above.
27225 verbose exit [ 1, true ]
`
Thanks for taking care.

RalphHogenbirk, 2017-07-04:
@Chris-Nbg thanks for posting, I was able to fix the problem and pushed it to develop (d1f4756). Let me know if it helps

Chris-Nbg, 2017-07-04:
Thanks @RalphHogenbirk - that did the trick!

kwmartin, 2017-07-05:
For info: previously I could not get PyMakr in Atom to work despite trying many different suggestions. Just came back to it (Ubuntu 16.04) and upgraded Atom to 1.18.0x64 and re-installed the PyMakr plugin. It's now (finally) working for me.

Ledbelly2142, 2017-07-26:
I tried to upgrade to the latest version 1.0.3 I think and it crashed Atom with similar stack dumps as above. Running Windows 10. Does not seem like its working for many. Just gonna try the new plugin for VS Code instead.

RalphHogenbirk, 2017-07-27:
@Ledbelly2142 It's a known issue on windows, specifically upgrading from 1.0.2 to 1.0.3. Uninstalling the plugin and re-installing it should solve the issue. But feel free to try the VS Code plugin anyway :)

Ledbelly2142, 2017-07-27:
@RalphHogenbirk 
Thanks, I did uninstall it and reinstall the plugin and had the same issue. The VS plugin works just fine. Electron does not seem to perform well on Win10 overall, just seems... VS code works for me on Win10, I'll just use that can chalk it up to node library versions not in sync (my guess, just looking for the least path of resistance). If I were using OSX or Linux, I would be using the plugin. Thanks for all the great work, it's appreciated.

RalphHogenbirk, 2017-07-27:
Ok, thanks for the kind words @Ledbelly2142. I'm still curious why the atom plugin doesn't work in your case. If you would ever get bored and mange to solve the problem with the atom plugin, let me know. In any case, enjoy the VS Code plugin :)

Nick-Aus, 2017-07-28:
I had this error too but got around it by creating a \packages\Pymakr\build\ directory and copying the appropriate version of serialport.node (Win32/64/etc) to it. This error existed with both the master and dev streams, manually running the install. All seems happy since I've put the file where it can be found.

RalphHogenbirk, 2017-07-28:
@Ledbelly2142 thanks a lot for the info, much appreciated. The weird thing is this part:
"(C:\Users\xxxxxx.atom\packages\Pymakr\lib\connections\serialport-win\lib\serialport.js:12:25)"
Which is a folder that was in 1.0.2 but has been completely removed from 1.0.3, including all references too it. But on windows, atom seems to have rembembered it and still tries to get it from that folder instead of from the /precompiles/ folder where the serialport files are located now

The solution from @Nick-Aus seems to be a valid workaround though! I can try to build that in to the setup script, placing the correct serialport.node in the build folder to make it work. But still I want find out why windows is searching in the wrong folder to begin with and starts searching for it in /build (among others). I'll investigate further and make a workaround or fix for it on monday.

rykelley, 2017-08-10:
I'm still having this issue. copied the serialport.node into the new build dir. and rebuilt. same issue. getting a little old.

markusc90, 2017-08-10:
It appears that thew newest version of atom (1.19) has broken the package again. It won't build serialport.node (because it says it was built with NODE_MODULE_VERSION 49 when it needs NODE_MODULE_VERSION 53), and when I try to build it from ~/.atom/packages/Pymakr/precompiles/serialport-win using "apm install" or "npm install" it cannot find the bindings file.

rbraggaar, 2017-08-11:
Running node install-win without node-modules directory gives me the following error:

```C:\Users\robbr\.atom\packages\Pymakr>node install-win
Installing serialport
Installing electron rebuild
Rebuilding...
{ Error: Command failed: C:\Users\devel\Projects\pymakr-atom\node_modules\.bin\electron-rebuild -f -w serialport -v 1.3.13
The system cannot find the path specified.

    at ChildProcess.exithandler (child_process.js:204:12)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at maybeClose (internal/child_process.js:891:16)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
  killed: false,
  code: 1,
  signal: null,
  cmd: 'C:\\Users\\devel\\Projects\\pymakr-atom\\node_modules\\.bin\\electron-rebuild -f -w serialport -v 1.3.13' }
Done!
```

RalphHogenbirk, 2017-08-11:
@markusc90 Indeed it has, I'm working on it. It needs to be rebuild against this new version of atom/electron (it does that in install.js, which is run when performing apm instal from the plugin folder). I'm running in to some new issues while rebuilding for this version. When I figure it out I'll doe a new release for it.

We'll also update the compatibility data for the plugin, so atom knows this plugin doesn't work yet with any future updates. That gives us a few days to rebuild and release a new version.

RalphHogenbirk, 2017-08-11:
@rbraggaar That's correct :) One of the comments in the file says it:
`// remember to change the path on line 21 to direct to your user folder`

If you do want to try to rebuild it yourself on windows, edit the install script value in package.json to perform install-win.js (after fixing the path inside it), then run apm install. That makes sure install-win.js is run using the same versioning as atom.

rbraggaar, 2017-08-11:
Ah, how could I miss that. Now following your instructions the command runs succesfully. Changed package.json to use install-win and ran apm install again.

RalphHogenbirk, 2017-08-11:
I've now uploaded a new version to develop with an atom 1.19.0 compatible precompile of the serialport library, but only for mac. For who had issues after updating to 1.19.0 on mac, please try it and let me know!

I've added it for win32 as well, but on my machine it is still causing issues finding the bind file (another issue that was there before). But for any windows users that want to try it, let me know what result it gives on your machine.

RalphHogenbirk, 2017-08-11:
@rbraggaar reading your last message, did you manage to rebuild it for windows and solve the issue?

rbraggaar, 2017-08-11:
Sorry, it did not recompile and solve the issue for me. Let me know if you want me to try something else

Ledbelly2142, 2017-08-12:
@RalphHogenbirk 
I was using VS Studio until I had issues with it... I wanted to look at more then one folder and work on code. Its a simple request, back to Atom. I dove into the dir where the serialport.node should be and lo and behold the /build directory was not there. Totally missing. I created the dir and reinstalled the Pymkr package and got  a new error: 

```
The module '.atom\packages\Pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

```

I'm not sure the difference between npm rebuild and npm install. So I did both and got a new error where a similar issue where the 'build' dir was missing from 

`.atom\packages\Pymakr\node_modules\bindings\build\serialport.node`

In addition these are missing as well:

```
\.atom\packages\Pymakr\node_modules\bindings\build\Debug\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\build\Release\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\out\Debug\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\Debug\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\out\Release\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\Release\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\build\default\serialport.node
\.atom\packages\Pymakr\node_modules\bindings\compiled\7.4.0\win32\x64\serialport.node
```

I tried to uninstall the Pymakr Package but couldn't with the \build dir I manually added. I deleted the package manually and am back to the first error.   

Now I am back to this error:

```
\.atom\packages\Pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
```

I have nuked the package dir in Pymakr and did an npm install... 

Still not working. I appreciate any help.

Thanks

saustin99, 2017-08-18:
Same here, on Windows 10. It looks like this problem's been active for months. Any ETA? I don't mean to sound snippy, but I'll just return it if there's no fix expected in the next week or so. Here's my error:
The module '\\?\C:\Users\Steve\.atom\packages\Pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

[38 more replies]