pycom/pymakr-atom #2: Connection timeout when device is in a loop

rbraggaar, 2017-04-13:
My Lopy is setup to read sensor values and print them in a loop.
Apparently this hinders the REPL console from recognizing the password option.

[image]

No problems in Putty.

RalphHogenbirk, 2017-04-14:
As we already discussed on the forum, this problem is not due to the device being in a loop, since signin and micropython run on 2 different threads. Ofcourse it could be that there is a bug in the plugin itself where it can end up in this state. 

Is it solved when you reboot both the device and atom? Any specific order of actions you could give to reproduce the problem, or does it already appear when connecting after startup?

danicampora, 2017-04-14:
Hi Rob,

Could you please share a pice of code running on the LoPy/WiPy that reproduces the issue. That way we can test it and solve it. Thanks!

Cheers,
Daniel

rbraggaar, 2017-04-14:
I now removed the code from the Lopy but the problem is still the same. 

My steps:
I power the Lopy
Connect to its WLAN
Open Atom (it will start connecting and fail)

Reopening Atom doesn't help, neither does resetting the Lopy (or both)
It connected once for me, on Atom startup yesterday whereafter it dropped the connection and refuses to connect again.

RalphHogenbirk, 2017-04-26:
@rbraggaar After some testing I found out that when connecting on telnet to the local, the nodejs code is too fast with sending the password to the board, after it received the boards request to send it. It only happens sometimes with my board, and connecting through the router works fine.

I just pushed a fix to the develop branch which you can test if you want. Let me know if this fixes your issue.

rbraggaar, 2017-04-30:
Now works good for me, tested on Wipy and Lopy.

RalphHogenbirk, 2017-05-01:
Good to hear! I'll to a merge to master soon

closed 2017-05-01 by RalphHogenbirk