pycom/pymakr-atom #3: REPL command prompt is outside of the console area

jenskueper, 2017-04-13:
I just installed Atom + the new Pymakr Plugin. Connection to the WiPy 2.0 module was successful but the command prompt is outside of the console area. I use Windows 8.1 x64 and my screen has a resolution of 1920*1200 (16:10). If you need any further information to tackle down this issue, do not hesitate to ask.

[image]

RalphHogenbirk, 2017-04-14:
Hi Jens, I've had one more internal report of this exact same issue. It's caused by the xterm library dynamically overriding the line-height property based on some calculation. With most users, the line height is 17px which makes it fit perfectly. Increasing the height to 223px solves the issue for users with this problem, but leaves the rest with a bar of extra space. 

I'll try to make a more dynamic solution for this problem today.

RalphHogenbirk, 2017-04-14:
Should be solved with the last commit on the develop branch (8db0960). @jenskueper if possible can you test this on your system to confirm?

jenskueper, 2017-04-14:
@RalphHogenbirk I almost expected that the fix wouldn't be that easy. I just installed the package from the develop brach and the bug is fixed on my system. Thank you very much for the quick fix!

closed 2017-04-14 by jenskueper

RalphHogenbirk, 2017-04-14:
Good to hear, thanks for testing! Will release a new version that includes the fix first thing monday morning.