pycom/pymakr-atom #6: Uncaught Error: EACCES: permission denied, scandir '/Volumes//Data'

robmarkcole, 2017-04-20:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.16.0 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.12.4
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 0.9.3

### Stack Trace

Uncaught Error: EACCES: permission denied, scandir '/Volumes//Data'

```
At fs.js:951

Error: EACCES: permission denied, scandir '/Volumes//Data'
    at Error (native)
    at Object.fs.readdirSync (fs.js:951:18)
    at Object.fs.readdirSync (ELECTRON_ASAR.js:571:28)
    at Sync._getFiles (/packages/Pymakr/lib/board/sync.js:217:15)
    at Sync._getFilesHashed (/packages/Pymakr/lib/board/sync.js:238:40)
    at Sync.start (/packages/Pymakr/lib/board/sync.js:54:28)
    at PymakrView.sync (/packages/Pymakr/lib/pymakr-view.js:324:18)
    at HTMLButtonElement.PymakrView.button_sync.onclick (/packages/Pymakr/lib/pymakr-view.js:61:15)
```

### Commands

```
```

### Non-Core Packages

```
Pymakr 0.9.3 
```

RalphHogenbirk, 2017-04-25:
HI @robmarkcole Do you have any content in the 'sync folder' setting?

RalphHogenbirk, 2017-05-17:
@robmarkcole did you eventually find out what was going wrong? Is it still an issue?

robmarkcole, 2017-05-17:
Hi Ralph
have updated Pymakr and no problems now, thanks

closed 2017-05-17 by RalphHogenbirk