pycom/pymakr-atom #7: Uncaught TypeError: _this.view.synchronize is not a function

lincolnsmithy, 2017-04-23:
[Enter steps to reproduce:]

1. ATOM:  Choose Menu Packages->Pycom Sync->Syncronize Code will generate the error.
Also note, when using sync "button" I always get unable to find folder message - Folder appears legitimate but never works.

**Atom**: 1.16.0 x64
**Electron**: 1.3.13
**OS**: Mac OS X 10.11.5
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 0.9.3

### Stack Trace

Uncaught TypeError: _this.view.synchronize is not a function

```
At /Users/john/.atom/packages/Pymakr/lib/main.js:21

TypeError: _this.view.synchronize is not a function
    at /packages/Pymakr/lib/main.js:21:45)
    at CommandRegistry.module.exports.CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:259:29)
    at CommandRegistry.handleCommandEvent (/app.asar/src/command-registry.js:3:59)
    at CommandRegistry.module.exports.CommandRegistry.dispatch (/app.asar/src/command-registry.js:160:19)
    at AtomEnvironment.module.exports.AtomEnvironment.dispatchApplicationMenuCommand (/app.asar/src/atom-environment.js:1179:28)
    at EventEmitter.outerCallback (/app.asar/src/application-delegate.js:312:25)
    at emitThree (events.js:116:13)
    at EventEmitter.emit (events.js:194:7)
```

### Commands

```
     -6:27.9.0 core:move-right (input.hidden-input)
     -5:47.8.0 core:backspace (input.hidden-input)
 11x -5:47.2.0 core:move-right (input.hidden-input)
     -5:44.8.0 core:backspace (input.hidden-input)
 11x -5:37.6.0 core:move-left (input.hidden-input)
     -5:35.9.0 core:move-right (input.hidden-input)
     -5:35.6.0 core:backspace (input.hidden-input)
     -5:00.7.0 tree-view:copy-full-path (span.name.icon.icon-file-directory)
     -4:57.7.0 core:paste (input.hidden-input)
     -2:26.8.0 pymakr:synchronize (ol.tree-view.full-menu.list-tree.has-collapsable-children.focusable-panel)
  2x -1:07.9.0 pymakr:toggleREPL (div.package-detail.panels-item)
     -0:51.1.0 pymakr:synchronize (div.package-detail.panels-item)
```

### Non-Core Packages

```
Pymakr 0.9.3 
```

RalphHogenbirk, 2017-04-25:
HI @lincolnsmithy 
Sorry for the late reply. The type error that caused this has been fixed in the code on the develop branch. It'll be included in the next release.

About the 'unable to find folder' message when synchronizing: what is the exact content in your 'Sync folder' setting?

lincolnsmithy, 2017-04-25:
I only had basic boot.py and main.py at this point.  I did however work around the problem (for now).
If you leave the Sync Folder blank it will use the first folder in the Atom Folder View (sorry not on machine with Atom so I may be calling it wrong.  So, I moved my sync folder above the .Atom folder and that seems to work.  Will keep an eye out for your update...
Thanks,
John

RalphHogenbirk, 2017-04-26:
Good to see you got it working @lincolnsmithy. If I understand correctly, your project folder was inside the .Atom folder? In that case it might explain the problem, since I don't think plugins are allowed to read anything inside the 'hidden' .Atom folder.  

I'll close this issue for now. If you run into the issue again after the next release, feel free to re-open this issue.

closed 2017-04-26 by RalphHogenbirk

lincolnsmithy, 2017-04-26:
Thanks, just for clarity the project folder was not in the .Atom folder...