pycom/pymakr-atom #11: Connection information is not stored per project

ButchAnton, 2017-04-27:
Connection information (IP address or serial port) should be stored and used per project, not globally.

If I have two LoPy boards talking to each other (in a gateway/node scenario, for example), I can't configure one Atom project to talk to the gateway and another to talk to a node.  This means I have to constantly swap ports around to upload code to the right board.  This is extremely error prone and time consuming.  :-(

danicampora, 2017-04-27:
I second this. @RalphHogenbirk please add this to the top of the list (right after the copy/paste)

RalphHogenbirk, 2017-04-28:
Hi @ButchAnton I agree as well. It's been in the list of improvements from the start (see 'planned improvements' in my [forum post](https://forum.pycom.io/topic/1026/pymakr-atom-plugin-released/39)).

RalphHogenbirk, 2017-05-16:
This feature has been added in release 1.0.0

closed 2017-05-16 by RalphHogenbirk