pycom/pymakr-atom #13: Sync only works once

theRealBugger, 2017-05-02:
I updated my Atom plugin and my LoPy firmware to the lastest versions. The project sync command works only the "first time", then the sync process hangs forever like this:
```
Syncing project (main folder)...

>>>
>>>
```

Then I need to reset the board, restart Atom and format the flash memory to make the sync work again (for just one time!!)
my project actually contains 15 files, 60kb total.

RalphHogenbirk, 2017-05-02:
It's possible that your project is using a significant amount of ram, leaving not enough for the sync feature to work properly (resulting in an error on the board). Can you put your board in [safe mode](https://docs.pycom.io/pycom_esp32/pycom_esp32/toolsandfeatures.html?highlight=safe%20mode#safeboot) before you sync and see if that works?

In any case, error handling in this situation has to be improved on the plugin side.

theRealBugger, 2017-05-02:
Nope, my main.py does nothing. If I power cycle the board, free ram is 64752 and sync does not work

RalphHogenbirk, 2017-05-02:
Thanks for the information. For debugging, you could keep a seperate telnet connection open to the board (using iterm or so) during sync and see if you get any errors or useful information back (assuming you are syncing over serial).

If you have the time, you could also enable logging (set 'logging_level' to 0 in ~/.atom/packages/pymakr/lib/config.js), restart pymakr, open the console (cmd-opt-i) and copy the console output during sync here.

theRealBugger, 2017-05-02:
I have this errors just opening Atom

```
file:///C:/Users/danie/AppData/Local/atom/app-1.16.0/resources/app.asar/static/xterm.css Failed to load resource: net::ERR_FILE_NOT_FOUND
C:\Users\danie\AppData\Local\atom\app-1.16.0\resources\app.asar\src\task.js:46 spell-check:en-us Can't load en_US: Cannot find dictionary for en_US.
```

And that's the output of the "sync" button:

```
[silly] Pyboard | Received message: 
>>> 
>>> 
logger.js:23 [silly] Pyboard | Received message: 
raw REPL; CTRL-B to exit
>
pymakr-view.js:122 Status: 3
pymakr-view.js:124 Writing extra line
logger.js:23 [silly] Pyboard | Received message: OK
logger.js:23 [info] Sync | Reading pymakr file
logger.js:23 [silly] Pyboard | Received message: 
```

bjoernbusch, 2017-05-02:
I have the same issue, after the first sync, it won't sync again. This is only happening since the update a couple days ago.
```
[silly] Pyboard | Received message: OK
[info] Sync | Reading pymakr file
[silly] Pyboard | Received message: �
[silly] Pyboard | Received message: [["boot.py","f","acf2e83ea1bcc0388282558c9643b9234aa4242c1c90e
[silly] Pyboard | Received message: 16e210f5e4fcdce483a"],["main.py","f","e3b0c44298fc1c149afbf4c8
[silly] Pyboard | Received message: 996fb92427ae41e4649b934ca495991b7852b855"]]
[silly] Pyboard | Received message: >
```
My boot.py contains very little data. I'm on the latest FW, it was working fine with the last version of the Atom plugin.

RalphHogenbirk, 2017-05-03:
Is this happening on serial, telnet or both? I haven't been able to reproduce it yet.

I suspect it's going wrong while reading the project file from the board (the json file that contains the file status). I've added improved error handling when reading the project file to the develop branch now (from commit #fe26110). It now displays a message 'failed to read project file' and continues synchronizing anyway. 

@bjoernbusch @theRealBugger Can one of you try the code on develop and tell me if you see this message?

theRealBugger, 2017-05-03:
Ok @RalphHogenbirk sync worked ok for a couple of hours (syncing only the modified files), then prompted this output (I did modified just one file in my project):
```
Failed to read project status, synchronizing all files
[1/12] Writing file communication.py
Synchronizing failed: Write failed. Please reboot your device manually.
```

And developer console shows these errors:
```
Failed to load resource: net::ERR_FILE_NOT_FOUND    C:\Users\danie\AppData\Local\atom\app-1.16.0\resources\app.asar\src\task.js:46 
spell-check:en-us Can't load en_US: Cannot find dictionary for en_US.
48index.html:1 Uncaught TypeError: undefined is not a function
(program):116 Handling of 'mousewheel' input event was delayed for 101 ms due to main thread being busy. Consider marking event handler as 'passive' to make the page more responive.
(program):677 Handling of 'wheel' input event was delayed for 101 ms due to main thread being busy. Consider marking event handler as 'passive' to make the page more responive.
75index.html:1 Uncaught TypeError: undefined is not a function
```

swfecsacSWse, 2017-05-11:
Today I upgraded to Pymakr plugin 0.9.4,  latest SiPy firmware, and latest PlatformIO software.  Sync then wouldn't work at all for me, producing the same screen output as the original post.   Reverting back to 0.9.3 worked fine.

theRealBugger, 2017-05-11:
With the upgrade on develop branch It works, but I have that new error sometimes.

RalphHogenbirk, 2017-05-11:
I've put some new bugfixes on develop just now. Solves the issue of not being able to read the project file but also the sync failing randomly. @theRealBugger @swfecsacSWse feel free to test it!

servicing, 2017-05-11:
Solved now for me.

RalphHogenbirk, 2017-05-16:
I'm assuming this is solved with the newest code on develop judging from @servicing's comment and the lack of other responses. If anyone still has this or a similar problem, feel free to re-open or create a new issue.

closed 2017-05-16 by RalphHogenbirk

jdcrunchman, 2019-07-29:
I'm using PyMakr 1.4.14.    For some reason,  the Synch is not working properly.
>>> import os
>>> os.listdir('www')
['css', 'fonts', 'img', 'index.html', 'p-contacts.html', 'p-msg.html', 'w-msg.html', 'test.pyhtml', 'www', 'MainSetup.py', 'Quickies', 'ideas.py', 'lora_api.py', 'project.pymakr', 'mainweb.py', 'microWebSocket.py', 'microWebSrv.py', 'microWebTemplate.py', 'quickies.py', 'user.py', 'webview.py', 'tst.py']
>>>

I am only supposed to have these files in my "www" directory....
['css', 'fonts', 'img', 'index.html', 'p-contacts.html', 'p-msg.html', 'w-msg.html', 'test.pyhtml']

But as you see,  most of the files in my root folder,  are also displayed.  As you see above,  another "www" folder was created.   

Then,  I tried to go into the 2nd "www" directory,  which I thought was somehow created.
I was already in the first "www" directory,  but wanted to try to go into the 2nd one.

>>> os.chdir('wwww')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 

When I did this,  I got an OSError,  so I'm confused.   

Do you think that these other files (WHICH APPEARED) to be added to the "www" filles as I've shown above,  have just come from the root,  but somehow displayed when I went into the "www" directory?

Could this be from a fuckup of the Synch function from PyMakr and be a bug.

From all I can tell,  there doesn't seem to be any "www" directory created into my original one.

I have trouble explaining things.
If any of the developers are interested in helping me,  or looking further,  please contact me at jdcrunchman@gmail.com,   or Signal or Telegram at +1 818-640-5290.

I have Zoom account,  and can screen share with you on Zoom or Skype (jdcrunchman).

I need to know if it's possible to re-synch the entire directory and file structure.    

Before,  I had to re-flash the ESP32 board,   re-install the PyMakr (which took me over 2 hours of struggle and failure),  I prefer not to have to do that again.

PLEASE HELP.

jdcrunchman, 2019-07-29:
Oh,  I did notice this,   and had entered this incorrectly...  My eyes are VERY BAD,  and for weeks now,  I've been trying to make the ESP32 console text larger so I can read it,    so far,  no luck,   I sure can use some pointers to change the font size.  I had originally put in....

os.chdir('wwww')

should have been 
os.chdir('www')

But I also observed some other interesting and VERY ANNOYING behavior

I often import or run a file I call Quickies,  which defines simple file management stuff.

One such function I defined was this...

from os import listdir, remove, chdir, getcwd
def pwd():
    getcwd()

Just to see what directory its in.

This function DOES NOT WORK....  But when I tried to copy the console output,
I get this...

Uncaught TypeError: Cannot read property 'length' of undefined
/Users/johndraper/.atom/packages/pymakr/node_modules/xterm/lib/utils/BufferLine.js:11
Show Stack Trace
The error was thrown from the pymakr package. This issue has already been reported.
View Issue

Which shows up when I try to copy the console,  so I can't show this to you.

I was able to copy the last part.
>>> cd('www')
>>> pwd() .     <---- notice here,  the pwd() function things it's root.
>>> import os  <---- so without using this function,  which for whatever,  didn't work.  
>>> os.getcwd()   <---- I imported the os,  and used directly,
'/www' .        <---- and presto,  there it is....

Is this a bug in the Micropython,  or a bug on PyMakr?    I have no clue.

Perhaps a bug on BOTH.

Can someone please contact me on Skype (jdcrunchman) so I can show you in detail,  and interact with you?

John