pycom/pymakr-atom #15: Reconnect button first click not working

theRealBugger, 2017-05-03:
To reproduce:
Open the console and make sure you're correctly connected using serial port;
press "reconnect" button and wait for the error message (it will appear); 
now click on the console to retry, or click "connect". It will connect instantly

RalphHogenbirk, 2017-05-05:
I got to reproduce this, but only with telnet, not with serial. I gave the board a little bit more time to actually disconnect before trying to reconnect, which fixed the issue. I'll push the fix to develop later today. It'll be in the next release.

RalphHogenbirk, 2017-05-05:
There you go, that should fix it. Try the code on develop to see if it works properly for you.

theRealBugger, 2017-05-05:
There's a problem installing from develop branch
[npm-debug.log.txt](https://github.com/pycom/pymakr-atom/files/978641/npm-debug.log.txt)

RalphHogenbirk, 2017-05-05:
Sorry about that, font-awesome lib had a wrong version number in package.json. Fixed it just now on develop (02b9cb7)

theRealBugger, 2017-05-05:
Nope, problem is still there. How can I help you further?

RalphHogenbirk, 2017-05-05:
Really, exactly the same error report (No compatible version found: font-awesome@^4.7.2)? Make sure you removed node_modules and try apm install or npm install again. If that doesn't help feel free to post the new npm log again so I can see what's going on.

theRealBugger, 2017-05-05:
Sorry, I installed succesfully. I meant the OP error is still there

RalphHogenbirk, 2017-05-05:
Stange, on my boards it works. You can try to increase the timeout in lib/connections/pytelnet.js:72 and see if that helps. Maybe 200ms is not enough for all setups.

RalphHogenbirk, 2017-05-16:
@theRealBugger do you still have issues with this? Or did an increased timeout work for you?

theRealBugger, 2017-05-16:
I have not tried yet but the problem is persisting

theRealBugger, 2017-05-19:
I changed timeout to 500ms but nothing is changed (I'm using v1.0.0 now). It doesn't look like a timing issue. It's like a variable is not toggled (if i double click the button very quickly the connection works)

RalphHogenbirk, 2017-12-19:
Old issue, I now, but this should be solved with the latest version, right? Have not heard anyone with this issue since. If still have it, please re-open the issue, but for now I'll close it.

closed 2017-12-19 by RalphHogenbirk