pycom/pymakr-atom #16: Sync write failed with particular file

theRealBugger, 2017-05-04:
It happened another time before now that synchronization always fails on a particular file. Changing the file content solved the problem.

Here's the console output:
```
[11/13] Writing file test.py                                                                                            
Synchronizing failed: Write failed. Please reboot your device manually. 
```

```
[info] Monitor | Seding write-file command for test.py
logger.js:23 [verbose] Monitor | Sending 16 bit name length 
logger.js:23 [verbose] Monitor | Sending name
logger.js:23 [verbose] Monitor | Sending 32 bit content length (283)
logger.js:23 [verbose] Monitor | Sending chunk 0 of 256
logger.js:23 [verbose] Monitor | Requesting ack...
logger.js:23 [verbose] Monitor | Got an ack back, flushing..
```

RalphHogenbirk, 2017-05-05:
Thanks for the extended info. I can't seem to reproduce it with that specific file, but I'm having the same problem in an unpredictable way now. I did find out it's a lot more stable when I disconnect the wifi of the board. It seems that some wifi activity can interfere with the sync code that's running on the board.

So i'm connected over serial, do a wl.disconnect() first (wl is where I have my WLAN object created in my boot file), then start syncing. 

Can you check if this solves the issue for you?

theRealBugger, 2017-05-05:
Wifi is in default AP state, so no network connected. If you intend to shut down wifi, how to do it?

Smeedy, 2017-05-14:
I'm syncing over serial on OS/X and having the same issues. Tried on 0.9.3 and 0.9.4 installed from a clone of the repository and give the same problem. Whenever a file gets too big, i have the sync error:

```
Type "help()" for more information.                                                                                     
>>>                                                                                                                     
Syncing project (main folder)...                                                                                        
                                                                                                                        
>>>                                                                                                                     
>>> Failed to read project status, synchronizing all files                                                              
Creating dir lib                                                                                                        
[1/4] Writing file boot.py                                                                                              
[2/4] Writing file lib/onewire.py                                                                                       
Synchronizing failed: Write failed. Please reboot your device manually. 
```

I've enclosed two zipfiles. One with a truncated lib/onewire.py file which does not give me any problem. And the one with the original lib/onewire.py which produces the console message shown. Also included are the Atom dev console snippets which have the relevant portions.

Other notes. Cable is an original FTDI ttl-232r-3v3 cable which gives no problem on flashing ESP32 from the IDF on 921k speeds.

Let me know if you need more info.

cheers,
Martijn

[lopy-test1_error.zip](https://github.com/pycom/pymakr-atom/files/999271/lopy-test1_error.zip)
[sync_error.txt](https://github.com/pycom/pymakr-atom/files/999273/sync_error.txt)
[lopy-test1_sync_ok.zip](https://github.com/pycom/pymakr-atom/files/999272/lopy-test1_sync_ok.zip)
[sync_ok.txt](https://github.com/pycom/pymakr-atom/files/999270/sync_ok.txt)

RalphHogenbirk, 2017-05-15:
@Smeedy Thanks for the extended info. Did you try the latest code of the develop branch? Last Thursday's commit (e98d2e4) was ment to fix exactly this issue.

Smeedy, 2017-05-15:
Yes I did. Latest development yesterday afternoon. Same issue.

theRealBugger, 2017-05-15:
@RalphHogenbirk Actually It seems that the problem is gone. I'll try again and let you know

Smeedy, 2017-05-15:
I'll check the e98d2e4 once more. Let you know later today, as I'm sure I've also ran the latest dev.

RalphHogenbirk, 2017-05-15:
@Smeedy I've tested your code and it failed with me as well. Turns out the length of the file, when converted to a 4 byte hex, happens to contain an escape character we're using, which was not filtered out properly. I've just pushed another commit to develop (3360b9c) which fixes the issue for me. 
Let me know if this works for you as well.

Smeedy, 2017-05-15:
Yes. This fix did indeed solve the issue. Can confirm it also worked on OS/X. Kudos.

Something probably not related, sometimes I have the a problem opening the `/dev/cu.usbserial-xxxxx` from within Atom, whereas running `screen /dev/cu.usbserial-xxxxxx 115200` (and properly closing screen) works without issues. Restarting Atom 1 out of 5 fixes this. If this persists I'll open a new ticket.

thanks again,
Martijn

RalphHogenbirk, 2017-05-16:
Good to hear! I'll close this issue now. 

I haven't heard of any issues with serial connection like that before, so I don't know what that could be. Making a new ticket if this persists is a good plan.

closed 2017-05-16 by RalphHogenbirk