pycom/pymakr-atom #17: Synchronization should have exclusion/inclusion list

ButchAnton, 2017-05-16:
There should be a way to have the sync functionality exclude files/directories.  For example, I have my source checked into git, and syncing the .git directory is really a bad idea.

RalphHogenbirk, 2017-05-17:
Have you tried to sync yet? It should already exclude any directory or file that starts with a . like the .git directory or the .gitignore file. It also ignores most filetypes, except for a short list (see general settings - sync file types). Let me know if it somehow behaves differently on your install.

Anyway it would be a nice addition to have something like a '.pyignore' file where you can say which specific files and directories to ignore. But it might not be needed with these current options. Let me know what you think about it.

ButchAnton, 2017-05-22:
Ralph, it syncs the whole .git directory.  It's not ignoring it.  It also doesn't seem to honor the sync file types directive.  Basically, it just sends everything over.

Also, it would be nice to be able to sync individual files.  For example, I just edited this one file and would like to push only it to the board.

danicampora, 2017-05-22:
OK, if it syncs the .git directoey that's a bug. Besides, it's supposed to only sync back the files that have changed, not the whole project every time. Ralph: can you check this? Thanks...

RalphHogenbirk, 2017-05-22:
@ButchAnton I don't have an expansion board at home at the moment so I can't test, but I'm pretty confident that those features are working. For sure the synchronizing of only the changed files in instead of all files. And [this line](https://github.com/pycom/pymakr-atom/blob/master/lib/board/sync.js#L236) prevents the .git folder to get sync'd

I'll test everything you are mentioning tomorrow first thing, but are you absolutely sure you are using the latest version? (v1.0.1 is out since today). Could it be that you manually installed an early version long ago, which might still be active?

ButchAnton, 2017-05-22:
@RalphHogenbirk I had version 1.0.0 installed and that's what was failing.  I've just updated to 1.0.1, and I'd really like to paste the output of clicking sync, but I can't figure out how to copy from the Atom window.  :-(

RalphHogenbirk, 2017-05-22:
@ButchAnton copying should work with ctrl-c or cmd-c from the terminal window itself. Which OS and version are you on?

ButchAnton, 2017-05-22:
I'm using MacOS 10.12.4, and selections do not persist for me to copy them.  If I select some text and then release the mouse button, the selection goes away -- no hope for command-c to copy.  I've tried command-c without releasing the mouse button, but that doesn't copy.

I'm just totally useless, aren't I?

ButchAnton, 2017-05-22:
OK, apparently copying is state dependent -- I have to be connected to the board for it to work.

'''
Syncing project (main folder)...                                                                                        
                                                                                                                        
Reading file status                                                                                                     
Failed to read project status, synchronizing all files                                                                  
Creating dir LoPy_BME280                                                                                                
Creating dir LoPy_BME280/.git                                                                                           
Creating dir LoPy_BME280/.git/hooks                                                                                     
Creating dir LoPy_BME280/.git/info                                                                                      
Creating dir LoPy_BME280/.git/logs                                                                                      
Creating dir LoPy_BME280/.git/logs/refs                                                                                 
Creating dir LoPy_BME280/.git/logs/refs/heads                                                                           
Creating dir LoPy_BME280/.git/logs/refs/remotes                                                                         
Creating dir LoPy_BME280/.git/logs/refs/remotes/origin                                                                  
Creating dir LoPy_BME280/.git/objects                                                                                   
Creating dir LoPy_BME280/.git/objects/93                                                                                
Creating dir LoPy_BME280/.git/objects/c0                                                                                
Creating dir LoPy_BME280/.git/objects/c7                                                                                
Creating dir LoPy_BME280/.git/objects/info                                                                              
Creating dir LoPy_BME280/.git/objects/pack                                                                              
Creating dir LoPy_BME280/.git/refs         
Creating dir LoPy_BME280/.git/refs/heads                                                                                
Creating dir LoPy_BME280/.git/refs/remotes                                                                              
Creating dir LoPy_BME280/.git/refs/remotes/origin                                                                       
Creating dir LoPy_BME280/.git/refs/tags                                                                                 
[1/4] Writing file LoPy_BME280/__init__.py                                                                              
Synchronizing failed: Write failed. Please reboot your device manually.                                                 
'''

As you can see, it decided that it needed to go grab a bunch of stuff that it shouldn't, though I don't know what it was (eventually) going to copy to the board, because that bit failed.  I tried rebooting it, but it has the same issue.

danicampora, 2017-05-22:
Copiny should work with the board disconnected. We need to look into that...

ButchAnton, 2017-05-22:
Sorry, I should be clear that I meant copying as part of copy/paste, not copying files to the board, which, I would think, should require one to be connected to the board.  :-)

RalphHogenbirk, 2017-05-22:
Ah, I get it now. You are trying to sync the complete Lopy_BME280 folder, which contains a git repo. My code assumes you have your git repository in the root of your project. I'm assuming this is your project folder itself, and you are not intending to create a LoPy_BME280 folder on your device? Either open this folder as a project in Atom, or change your sync_folder setting to match it. Then it should work.

The copy paste problem is probably because there is still input being printed from the device to the terminal, which means your selection will disapear. For now, stop your code from running with ctrl-c first before you copy. I'll check tomorrow if I have the same issue and if I can change this behaviour

danicampora, 2017-05-22:
Ralph: let's also try to ignore files like .git even when they are not in the place where your code assumes it. Thanks :-)

RalphHogenbirk, 2017-05-22:
Agreed @danicampora

ButchAnton, 2017-05-22:
The reason for the LoPy_BME280 being its own repository is that it's a library for the BME280.  I merely cloned the repo into my project and import it.

As for the copy/pasta problem, the board is not spewing output to the terminal.  I've already hit ctrl-c before I try to do the copy.

RalphHogenbirk, 2017-05-23:
@ButchAnton I've just pushed a new commit to develop. This might solve your sync problem. It doesn't sync any hidden sub-folders or empty folders anymore. Feel free to download it from develop and test it. 

I've tested the copy paste in the atom terminal and it's throwing away the selection as soon as the board prints anything to the terminal. I haven't found any option in the terminal lib to fix that yet.

ButchAnton, 2017-05-23:
@RalphHogenbirk That seems to have improved things.  Thank you!

RalphHogenbirk, 2017-05-29:
@ButchAnton Is it all solved now and can I close this issue? Or do you need more help with this?

ButchAnton, 2017-06-02:
I think it's working ok.  I'll reopen if I see further issues.

Thanks!

closed 2017-06-02 by ButchAnton