pycom/pymakr-atom #21: Project device address setting does not override global device address setting

ButchAnton, 2017-05-23:
If one specifies a device address in the project specific settings file (pymakr.conf), it does not override the device address setting in the global settings.

```
{
    "address": "LoRaGatewaySAP",
    "username": "micro",
    "password": "python",
    "sync_folder": ""
}
```
where one finds, in /etc/hosts/
```
192.168.0.107 LoRaGatewaySAP
```
and where one finds that it properly resolves
```
% ping LoRaGatewaySAP
PING loragatewaysap (192.168.0.107): 56 data bytes
64 bytes from 192.168.0.107: icmp_seq=0 ttl=255 time=10.451 ms
64 bytes from 192.168.0.107: icmp_seq=1 ttl=255 time=8.898 ms
```
the connection attempt fails, instead trying to connect to the default address:
```
Connecting on 192.168.4.1...
```

RalphHogenbirk, 2017-05-24:
Did you press 'reconnect' once after saving the config file? Is it only the 'address' setting that's not working from the config file, or do none of them work?

ButchAnton, 2017-05-24:
@RalphHogenbirk I tried that yesterday without success.  However, after shutting everything down for the night, when I fired everything back up this morning, everything worked correctly.  I guess it just needed a night to sleep on it and consider its behavio(u)r.

RalphHogenbirk, 2017-05-29:
@ButchAnton Did the bug decide to come back in the last 5 days? Or was it a one time occurrence?

ButchAnton, 2017-06-02:
Seems to be working ok.  Thanks!

closed 2017-06-02 by ButchAnton