pycom/pymakr-atom #23: Pymakr settings fails will not show in Atom

danwalkowski, 2017-05-29:
Fails to show Pymakr settings  in Atom 1.17.2 (Mac).
Cannot see or change any settings.

RalphHogenbirk, 2017-05-31:
Hi @danwalkowski, thanks for the feedback. I spotted a bug that is causing this. If you open atom and then open a project, it doesn't refresh the variable that stores the project path. 
I've now fixed this in the commit 8fc5a96 on develop and also added a proper error message when you try to open project settings without there being a project open.
You can get the code from develop and test this is you want. I'll include it in the next release ofcourse.

danwalkowski, 2017-06-01:
Thanks!  I believe I've found another issue though:
Now that the settings are being shown, connecting to the board always times out.  Could this be related to the extra long time it takes a Mac to get a reverse-lookup connection?
And is there a setting to change the timeout?

RalphHogenbirk, 2017-06-08:
If you connect to 192.168.4.1, the mac reverse-lookup indeed messes things up sometimes. Telnet connection from the commandline also takes forever and the atom timeout triggers sooner. This timeout is not configurable at the moment, though you can customize it in [/lib/settings-wrapper.js line 44 ](https://github.com/pycom/pymakr-atom/blob/master/lib/settings-wrapper.js#L44) if you'd really want to.

RalphHogenbirk, 2017-06-08:
I'll close this issue for now. Feel free to create a new issue for the connection timeout if needed.

closed 2017-06-08 by RalphHogenbirk