pycom/pymakr-atom #24: Can't install Pymakr-atom with Atom 1.17.2

davysuisse, 2017-06-01:
Is it a known issue? I'm trying to install the package, but it fails. 

ymakr 1.0.1

serialport-linux@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node
build@4.0.7 – Cannot find module 'C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build'
serialport-linux@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node
serialport-osx@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node
serialport-osx@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node
serialport-win@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node
serialport-win@4.0.7 – %1 is not a valid Win32 application.
 \\?\C:\Users\Davy Claude\.atom\packages\Pymakr\lib\connections\serialport-linux\build\Release\serialport.node

bnjroos, 2017-06-02:
Same problem here on windows 10. I have tried to install it manually also without success.

RalphHogenbirk, 2017-06-02:
@davysuisse @bnjroos I've seen this error once as well and the reason was that I manually linked multiple versions of the atom plugin to atom (using 'apm link' from a manually downloaded version). Could that be the case here? Otherwise I have some testing to do on windows.

bnjroos, 2017-06-02:
I used npm link, however I have only one link (it seems):
C:\Users\Benjamin\.atom\packages\Pymakr>npm ls -g Pymakr
C:\Users\Benjamin\AppData\Roaming\npm
`-- Pymakr@1.0.1  -> C:\Users\Benjamin\.atom\packages\Pymakr

davysuisse, 2017-06-03:
I make it work by uninstalling Atom 1.17 and installing a previous version (1.16).

bnjroos, 2017-06-06:
Same here, I made several tests so I did not knew if it was rolling back 1.16 that made it work however I forgot to untick the automatic updates and it update to 1.17.2 by itself and it is not working again.

RalphHogenbirk, 2017-06-07:
I'm looking in to this, and it seems that there is an issue with the bindings of the precompiled serialport library for windows. At least, I could only recreate the 'Could not locate the bindings file' error, not the ' %1 is not a valid Win32 application' error. 

Running 'npm install' from inside `~/.atom/packages/Pymakr/lib/connections/serialport-win/` solves it for me. I'll commit a permanent fix for this tomorrow.

@bnjroos since you also have this bindings problem and let me know if it patches the error.
@davysuisse Maybe this works for you too, otherwise I'll search for a win32 machine for me to test on

bnjroos, 2017-06-12:
@RalphHogenbirk I just tried without issue, now rolling back to 1.16 again :s

RalphHogenbirk, 2017-06-12:
@bnjroos I pushed a fix to development yesterday, sorry for not mentioning this. You can try this code as well if you don't mind switching again! Would appreciate the feedback. Make sure to run apm install again.

bnjroos, 2017-06-12:
@RalphHogenbirk Just tried it unfortunately it did not change anything :(

```
Failed to activate the Pymakr package
Could not locate the bindings file. Tried:
 → C:\Users\Benjamin\node_modules\bindings\build\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\build\Release\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\Debug\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\out\Release\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\Release\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\build\default\serialport.node
 → C:\Users\Benjamin\node_modules\bindings\compiled\6.5.0\win32\x64\serialport.node
```

RalphHogenbirk, 2017-06-12:
@bnjroos are you absolutely sure the apm install succeeded and the downloaded code from the develop branch is actually being run by atom? It looks like it searches for the 'bindings' lib in your global node modules folder instead of inside the lib within pymakr. That indicates that 'bindings' is not installed in either the main lib folder or the precompiled lib folder (where it should be installed)

bnjroos, 2017-06-12:
I've downloaded the develop branch and replaced the previous version, I did not use apm install but npm install which should also work isn't it?

RalphHogenbirk, 2017-06-12:
It's safer to use apm install, since that uses the internal node version that atom uses as well, while npm depends on your local modules.

bnjroos, 2017-06-13:
Just tried with apm, unfortunately it still encounter the issue.

RalphHogenbirk, 2017-06-27:
The latest code on the develop branch will solve this issue, both on atom 17.2 and 18.0. All precompiled serialport lib's have been updated for all operating systems.

Feel free to test this code @bnjroos @davysuisse and let me know if it works! I'll make a new release with this code soon.

bnjroos, 2017-07-06:
Still the same issue, I will have to downgrade again if I want to work on the board :s

bnjroos, 2017-07-06:
I tried the 1.0.3 and the developer branch with the 18.0

> Error: Could not locate the bindings file. Tried:
>  → C:\Users\Benjamin\node_modules\bindings\build\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\build\Debug\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\build\Release\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\out\Debug\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\Debug\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\out\Release\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\Release\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\build\default\serialport.node
>  → C:\Users\Benjamin\node_modules\bindings\compiled\6.5.0\win32\x64\serialport.node
>     at bindings (C:\Users\Benjamin\node_modules\bindings\bindings.js:88:9)
>     at Object.<anonymous> (C:\Users\Benjamin\node_modules\serialport\lib\bindings.js:3:35)
>     at Module.get_Module._compile (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
>     at Object.value [as .js] (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)
>     at Function.Module._load (module.js:424:3)
>     at Module.require (file:///C:/Users/Benjamin/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
>     at require (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
>     at Object.<anonymous> (C:\Users\Benjamin\node_modules\serialport\lib\serialport.js:12:25)
>     at Module.get_Module._compile (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
>     at Object.value [as .js] (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)
>     at Function.Module._load (module.js:424:3)
>     at Module.require (file:///C:/Users/Benjamin/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
>     at require (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
>     at Object.<anonymous> (file:///C:/Users/Benjamin/.atom/packages/pymakr/lib/connections/pyserial.js:20:18)
>     at Module.get_Module._compile (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
>     at Object.value [as .js] (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)
>     at Function.Module._load (module.js:424:3)
>     at Module.require (file:///C:/Users/Benjamin/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
>     at require (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:66:33)
>     at Object.<anonymous> (file:///C:/Users/Benjamin/.atom/packages/Pymakr/lib/board/pyboard.js:3:34)
>     at Module.get_Module._compile (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\native-compile-cache.js:106:36)
>     at Object.value [as .js] (C:\Users\Benjamin\AppData\Local\atom\app-1.18.0\resources\app\src\compile-cache.js:239:29)
>     at Module.load (module.js:473:32)
>     at tryModuleLoad (module.js:432:12)

RalphHogenbirk, 2017-07-07:
@bnjroos Sure you used 'apm install' instead of 'npm install' ? If so, does running 'node install.js' manually, result in any errors?

JoostHazelzet, 2017-08-09:
I got this issue today too on Windows 10 Home 64 bit. I noticed that Atom was recently automatically upgraded to 1.19.0 (release date 8 Aug). I disabled the auto upgrade function (Help - About Atom), next removed Atom and reinstalled 1.18.0 version. It did resolve the issue and Pymaker plug-in is working again. I keep the auto upgrade disabled until a permanent fix is delivered.

closed 2017-12-19 by RalphHogenbirk

vigneswaranjaga, 2019-07-30:
Hi Could someone help me with this? I have tried many methods but getting the same error with it.?

PC\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).