pycom/pymakr-atom #25: Failed to activate the Pymakr package

anithp, 2017-06-13:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.17.2 x64
**Electron**: 1.3.15
**OS**: Microsoft Windows 10 Home
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.0.1

### Stack Trace

Failed to activate the Pymakr package

```
At Could not locate the bindings file. Tried:
 → C:\Users\ANITH\node_modules\bindings\build\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\out\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\default\serialport.node
 → C:\Users\ANITH\node_modules\bindings\compiled\6.5.0\win32\x64\serialport.node

Error: Could not locate the bindings file. Tried:
 → C:\Users\ANITH\node_modules\bindings\build\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\Debug\serialport.node
 → C:\Users\ANITH\node_modules\bindings\out\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\Release\serialport.node
 → C:\Users\ANITH\node_modules\bindings\build\default\serialport.node
 → C:\Users\ANITH\node_modules\bindings\compiled\6.5.0\win32\x64\serialport.node
    at bindings (~/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (~/node_modules/serialport/lib/bindings.js:3:35)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at Object.<anonymous> (~/node_modules/serialport/lib/serialport.js:12:25)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at /packages/Pymakr/lib/connections/pyserial.js:14:18)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.17.2/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:66:1)
    at /packages/Pymakr/lib/board/pyboard.js:3:34)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.17.2/resources/app/src/native-compile-cache.js:106:1)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.17.2/resources/app/src/compile-cache.js:230:1)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
```

### Commands

```
```

### Non-Core Packages

```
atom-clock 0.1.7 
atom-python-debugger 0.3.4 
Pymakr 1.0.1 
```

RalphHogenbirk, 2017-06-13:
Similar issues are going on in [this issue](https://github.com/pycom/pymakr-atom/issues/24). 

Can you try to run the code that is on the development branch right now, and run 'apm install' from the package folder? I'm curious if this works for you. For me it did, for bnjroos it didn't.

anithp, 2017-06-13:
What is apm install?

RalphHogenbirk, 2017-06-13:
It's a command that does the same as when you install a new plugin in atom, and installs all dependencies. Check out https://github.com/pycom/pymakr-atom/tree/develop#manual-install for more info :)

anithp, 2017-06-13:
ok got, but they say to install manually.

RalphHogenbirk, 2017-07-06:
@anithp can you confirm this is fixed with the latest release of 1.0.3?

bernienor, 2017-07-21:
I have similar issues With pymakr 1.0.3 on Atom 1.18. Upgraded from 1.0.2 (working fine).

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.0.3

### Stack Trace

Failed to load the Pymakr package

```
At Could not locate the bindings file. Tried:
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\out\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\out\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\default\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\compiled\6.5.0\win32\x64\serialport.node

Error: Could not locate the bindings file. Tried:
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\out\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\Debug\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\out\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\Release\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\build\default\serialport.node
 → C:\Users\hembr_000\.atom\packages\Pymakr\compiled\6.5.0\win32\x64\serialport.node
    at bindings (/packages/Pymakr/node_modules/bindings/bindings.js:88:9)
    at /packages/Pymakr/lib/connections/serialport-win/lib/bindings.js:3:35)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.18.0/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/Pymakr/lib/connections/serialport-win/lib/serialport.js:12:25)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.18.0/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/connections/pyserial.js:20:18)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.18.0/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
    at Function.Module._load (module.js:424:3)
    at Module.require (~/AppData/Local/atom/app-1.18.0/resources/app/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/board/pyboard.js:3:34)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.18.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.18.0/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:473:32)
    at tryModuleLoad (module.js:432:12)
```

### Commands

```
```

### Non-Core Packages

```
Pymakr 1.0.3 
```

bernienor, 2017-07-21:
Is there a quick way of degrading to 1.0.2?

RalphHogenbirk, 2017-07-21:
Hi @bernienor, there is a problem with updating from 1.0.2. to 1.0.3 on windows, where some of the files are not updated. Completely uninstalling and re-installing again should solve the issue for now.
(for future reference: If you want to downgrade you can do `apm install Pymakr@1.0.2`)

bernienor, 2017-07-21:
How do you define competely uninstall. I've uninstalled both the pymakr and Atom itself. Are there some directories I need to delete as well?_

RalphHogenbirk, 2017-07-21:
I ment only the pymakr plugin itself. If you want to really make sure, check if the `~/.atom/pymakr/packages/Pymakr` folder has been completely deleted before installing the plugin again.

RalphHogenbirk, 2017-07-24:
To follow up on this: is your problem solved now @bernienor?

bernienor, 2017-07-30:
Sorry, been on vacation untill now.

Installing Python 2.7 did the trick. I have 3.6 as standard on my machine. Just got it running. Now back to debugging my code. Thanks a lot for the inputs.

bernienor, 2017-07-30:
For future reference check python version in use by the command:
`apm version`
Python must be 2.7

RalphHogenbirk, 2017-07-31:
That's good to hear @bernienor. I didn't know the python version difference could cause this error, so thanks for that info! I have some unfinished code laying around that indeed uses apm version to check the python version during installation. Seems like it's time to finish that up.

bobpotter3999, 2017-08-11:
@RalphHogenbirk 

Hi Ralph,

I don't know if you can help but you seem very active in this area. I had a LoPy working on Pymakr package 1.03 on a Windows10 version of Atom. The setup was just about usable then something changed as today I now get:-

The module '\\?\C:\Users\Arqiva\.atom\packages\Pymakr\precompiles\serialport-win32\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

I've seen an e-mail from Markus Soursa with exactly the same problem. Do you happen to know if there is there is a fix likely to come out in the near future?  Markus thinks it's after an update to Atom. As I never knowingly updated Atom is this something that could happen periodically? If there is an auto-updating feature in Atom is this something that's easy to disable; as it's a bit disconcerting when a set up that was working just stops without actually doing anything :)

Thanks if you can help but no worry if you can't.

kind regards

Bob Potter (Arqiva)

RalphHogenbirk, 2017-08-11:
@bobpotter3999 Correct, it's because of the atom update to 1.19.0 (it updates automatically by default). We're working on a solution, also to prevent it from breaking on some pc's after any future update.

The problem is with the serialport library, which doesn't install and build properly on all pc's. So there is a pre-compiles version of the library included for those cases. Unfortunately this one needs to be re-build when an update occurs. I'm working on rebuilding it now, but it is giving me some issues on windows.

bernienor, 2017-08-13:
I just got the same error message as bobpotter3999. Was not aware of the automatic update setting. It's turned off now. Very frustrating to once again having to debug the development tools.

bernienor, 2017-08-13:
Here's my workaround:

First I turned off automatic updates in Atom. 

In my .../AppData/Local/Atom folder I had two Atom install folders:
`app-1.18.0
app-1.19.0`

I simply removed (renamed) the app-1.19.0 folder and restarted Atom. 

Now Atom will start up in version 1.18.0.

Another way if you don't have the 1.18.0 folder is to download the zip file of the atom release of your desire.  [https://github.com/atom/atom/releases](Atom releases) I've not tested this my self

rakuna, 2017-08-14:
Had the same issue (auto-update to 1.19.0 breaking the loading of the pymakr plugin).
Was also able to get it to work by uninstalling atom, downloading and installing the [(1.18.0)](https://github.com/atom/atom/releases/tag/v1.18.0) release and disabling auto-updates in atom.
Looking forward to a solution that will allow auto-updates to be re-enabled in atom.

RalphHogenbirk, 2017-09-01:
Fixed in the just released version of 1.1.0. Don't forget to update to 1.19.0 as well, because it might still occur on 1.18.0!

closed 2017-12-19 by RalphHogenbirk

manibenwa, 2018-03-14:
Hai,
Iam using 1.24.1 x64 ATOM and iam trying to install the pymakr plugin in my windows 7 it showing the following errors
nstalling “pymakr@1.2.9” failed.Hide output…

npm ERR! Windows_NT 6.1.7600
npm ERR! argv "C:\\Users\\user\\AppData\\Local\\atom\\app-1.24.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\user\\AppData\\Local\\atom\\app-1.24.1\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\user\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\user\\.atom\\.apmrc" "install" "C:\\Users\\user\\AppData\\Local\\Temp\\d-118214-3780-svg4vw.9lwpmhd7vi\\package.tgz" "--runtime=electron" "--target=1.6.16" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! file C:\Users\user\.atom\.apm\debug\2.6.9\package\package.json
npm ERR! code EJSONPARSE

npm ERR! Failed to parse json
npm ERR! Unexpected token 'r' at 1:1
npm ERR! ric_1 = require("./Generic");
npm ERR! ^
npm ERR! File: C:\Users\user\.atom\.apm\debug\2.6.9\package\package.json
npm ERR! Failed to parse package.json data.
npm ERR! package.json must be actual JSON, not just JavaScript.
npm ERR! 
npm ERR! This is not a bug in npm.
npm ERR! Tell the package author to fix their package.json file. JSON.parse

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\user\AppData\Local\Temp

Please help how can i fix it?????