pycom/pymakr-atom #26: Connection to LoPy (through Pysense) stops

jellium, 2017-06-23:
Hello,

Using a LoPy connected through a Pysense board, once the REPL prompt over serial is reached, the connection stops after a few seconds and Pymakr console outputs the following error:
```
> Failed to connect (Error: Error: Broken pipe, cannot drain). Click here to try again.
```

This error is not reproduced when reaching the REPL prompt with `minicom`.

This error is not reproduced in Pymakr when using an expansion board instead of a Pysense board.

Thank you.

jellium, 2017-06-23:
Hi again,
I suspect the error to be related to the latest release of the Pymakr package for Atom (version 1.0.2). I am pretty sure this error occurs since I updated the package yesterday.

jellium, 2017-06-23:
I confirm that the problem lies in version 1.0.2 of the Pymakr Atom package.
[Downgrading to version 1.0.1](https://github.com/pycom/pymakr-atom/releases/tag/v1.0.1) solves the issue.

RalphHogenbirk, 2017-06-26:
hi @jellium, thanks for reporting this. I'm investigating what is causing this. 

On which OS are you? From 1.0.2, I added a keep-alive for the serial. [An issue](https://github.com/EmergingTechnologyAdvisors/node-serialport/issues/1024) in the serialport library could be the cause of this keep-alive not working on windows 10 specifically. Yet, on my virtual windows 10 machine it seems to be working using a pysense and wipy.

tobru, 2017-06-26:
I'm facing the same issue on Arch Linux, so it doesn't seem to be a Windows 10 specific issue.

jellium, 2017-06-26:
Hi @RalphHogenbirk, thanks for the answer.
I am using Ubuntu (16.04).

RalphHogenbirk, 2017-06-26:
Before this 'broken pipe' error appears, are you able to use the REPL for a few seconds? Or is it never showing the '>>>'?

jellium, 2017-06-26:
I confirm that I can use the REPL for a few seconds (it does show the '>>>' before the error occurs). I tried to explain this in the opening post.
Thanks,

RalphHogenbirk, 2017-06-26:
After some more testing I got to reproduce it on windows and linux. There is an issue with the driver in the pysense/pytrack that makes the keep-alive message fail. On mac it should also fail, but the drivers on mac are probably ignoring the error that is produced, so there it ends up working like it should, from atoms perspective.

We'll brainstorm about a solution. At the very least I'll have workaround for you guys by the end of the day.

jellium, 2017-06-26:
Sweet, thanks for your feedback.
Again: downgrading to 1.0.1 can be a workaround (it is for me in the meantime before any fixing update).

RalphHogenbirk, 2017-06-27:
@jellium @tobru A small workaround is now on the develop branch, that disables the keep-alive (dtr) for linux and windows. You can either keep using v1.0.1 or the code on develop. The workaround will be included in the next release. We'll have to wait for a future pytrack/pysense usb driver update before we enable the dtr feature again.

closed 2017-06-27 by RalphHogenbirk