pycom/pymakr-atom #28: font-awesome.less wasn't found

pschaus, 2017-07-05:
I copied pasted the latest version from the git commit 783ed90e612167ad39c5879a1eae97f40ef91f26

**Atom**: 1.18.0 x64
**Electron**: 1.3.15
**OS**: Mac OS X 10.12.5
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.0.2

### Stack Trace

Failed to load the Pymakr package

```
At '../node_modules/font-awesome/less/font-awesome.less' wasn't found. Tried - /Users/pschaus/.atom/packages/Pymakr/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/one-dark-syntax/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/one-dark-ui/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/static/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/font-awesome/less/font-awesome.less,../node_modules/font-awesome/less/font-awesome.less in /Users/pschaus/.atom/packages/Pymakr/styles/pymakr.less:7:0

LessError: '../node_modules/font-awesome/less/font-awesome.less' wasn't found. Tried - /Users/pschaus/.atom/packages/Pymakr/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/one-dark-syntax/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/one-dark-ui/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/static/node_modules/font-awesome/less/font-awesome.less,/Applications/Atom.app/Contents/Resources/app/node_modules/font-awesome/less/font-awesome.less,../node_modules/font-awesome/less/font-awesome.less
  at /packages/Pymakr/styles/pymakr.less:7:0
```

### Commands

```
     -2:17 core:copy (atom-notification.error.icon.icon-flame.native-key-bindings.has-detail.has-close)
```

### Non-Core Packages

```
Pymakr 1.0.2 
```

RalphHogenbirk, 2017-07-06:
Hi @pschaus, can you browse the node_modules folder and check if the `font-awesome/less/font-awesome.less` is there or not? Perhaps something else went wrong during install that prevented it from installing all dependencies. First, try with the latest release from today (v1.0.3) and see if that solves it.

pschaus, 2017-07-06:
Thank you @RalphHogenbirk The new release v1.0.3 fixed the problem.

closed 2017-07-06 by pschaus