pycom/pymakr-atom #30: OSX: Copy from console and paste into editor does not work

rolandvs, 2017-07-26:
If I use `atom` in combination with the `pymakr-atom` tool (nice work b.t.w) I cannot copy from the console to the editor or anything else. Copy from the editor and pasting into the console works fine.

Using MacOSX `10.12.5`.

RalphHogenbirk, 2017-08-04:
You are right, I'm having the same issue. The function that returns the selection (window.getSelection()) doesn't return anything. Possibly broke after atom 1.16, because with that version it was working. I'll look in to this!

RalphHogenbirk, 2017-08-28:
Sorry @rolandvs, not yet, because of the holidays. Now there are a few more pressing issues to be solved with the newer atom versions, but we'll get to it after that!

rolandvs, 2017-11-20:
@RalphHogenbirk Any news? It still doesn't work.

semireg, 2017-12-30:
Same issue here.

RalphHogenbirk, 2018-01-02:
@rolandvs It had been fixed two weeks ago but not published yet. So I just published it as a patch today (v1.2.7). 
Solution turned out to be quite simple once I saw that the xterm lib also keeps track of the selected text, which is now used instead of the windows object or any atom api.

closed 2018-07-06 by sdaniel55