pycom/pymakr-atom #31: Plugin terminal problem

leogaggl, 2017-07-29:
When installing the Pymakr plugin with Visual Studio Code the terminal window starts flashing between the standard command prompt and the following error message as soon as you enable the plugin:

[image]

**System info**
Visual Studio Code:
    Version 1.14.2
    Commit cb82febafda0c8c199b9201ad274e25d9a76874e
    Date 2017-07-19T23:26:08.116Z
    Shell 1.6.6
    Renderer 56.0.2924.87
    Node 7.4.0

Operating System:
    Ubuntu 17.04 64bit
    4.10.0-28-generic #32-Ubuntu SMP Fri Jun 30 05:32:18 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

RalphHogenbirk, 2017-08-01:
Hi @leogaggl, 

For future reference: please post vscode related issues in the pymakr-vsc repo and not the atom one. 

I've upgraded vsc on my ubuntu machine to the same versions as you are running, but everything seems to be working fine. The specific error you're getting means it can't open the js file that it has to execute for the terminal (terminalExec.js in the main plugin folder). Possibly something is wrong with the permissions to this file for some reason. Can you check if you have sufficient permissions enabled for vscode to access `~/.vscode/extensions/terminalExec.js`?

leogaggl, 2017-08-08:
Apologies @RalphHogenbirk - didn't see that there was a VSC repo !

It appears that ~/.vscode/extensions/terminalExec.js does not actually exist. I tried to replicate this on another system (Ubuntu 16.04LTS 64bit) with no Visual Studio Code installed and the issue was exactly the same:

`> umake ide visual-studio-code`

Install pymakr plugin via Extensions. Same problem as outlined above. What extension creates ~/.vscode/extensions/terminalExec.js ? The folder permissions are fine.

RalphHogenbirk, 2017-08-08:
@leogaggl I'm sorry, I see I made a mistake and forgot one folder in that path. The correct file location should be:
'~/.vscode/extensions/pycom.pymakr-0.1.1/terminalExec.js'

leogaggl, 2017-08-10:
Thanks @RalphHogenbirk 

-rwxr-xr-x terminalExec.js - owned by correct user and group. Does not seem to have any permission problem - everything appears fine.

gmires, 2017-08-14:
Hello,
I have the same problem with the same version of ubuntu
Permits and owners are ok !!!!!

News???

thank's
regards

gmires, 2017-08-15:
the solution is
install last nodejs by
 
## https://nodejs.org/en/download/package-manager/

Regards

gmires, 2017-08-16:
....and add usb device udev rules access by user.
Bye
Regards

leogaggl, 2017-08-17:
Hi @gmires 

Thanks for the suggestions. I installed node.js (via umake) and the flickering of the Pycom Console has indeed gone away ! However don't get any prompt or anything - just empty screen.

However I am still unable to connect or do anything in the Pycom console. Have you got some more detail on your udev rules. I addded a rule for the ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)

gmires, 2017-08-17:
Hello @leogaggl ,
I have the LoPy oem version ...
I connected to it a ftdi ttl / rs323 converter and created the following rule
In the /etc/udev/rules.d/ folder

Filename: 40-persistent-ftdi.rules

`ATTRS{idVendor}=="0403",ATTRS{idProduct}=="6001",MODE="0666",GROUP="users"`

Assigning the device to the user group and enabling read and write permissions.
You can see information about the device in dmesg, for example
Idvendor, idproduct, etc ..

exucuse for my easy english
regards

leogaggl, 2017-08-19:
Hi @gmires 

Thanks a lot - have the following:

SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="1615", GROUP="users:q
", MODE="0666"

Bus 002 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)

Will have some more play.

leogaggl, 2017-08-21:
Hi @gmires 

Thanks a lot - have the following:

SUBSYSTEMS=="usb", ATTRS{idVendor}=="0403", ATTRS{idProduct}=="1615", GROUP="users:q
", MODE="0666"

Bus 002 Device 004: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)

After some playing around I have got it working now as well. But I can only connect via WIFI. I have to connect to the LoPy access point and then the connection works. 

@RalphHogenbirk - is the dependency with Node.JS documented somewhere ? I could not see it anywhere.

leogaggl, 2017-08-21:
Closing this as I have moved over to the proper repo: [https://github.com/pycom/pymakr-vsc/issues/2](https://github.com/pycom/pymakr-vsc/issues/2)

closed 2017-08-21 by leogaggl

gmires, 2017-08-21:
hi @leogaggl ,
This way i connect you to the lopy via ssh eth.
Have you changed the configuration of pymakr plugin to make sure you do not connect to an ip address but to the serial device ??

(ex. /dev/ttyUSB0) in pymakr general option or project option.
In the box where at this time there should be the lopy ip address

regards

gmires, 2017-08-21:
hi@leogaggl,
see this:

[https://forum.pycom.io/assets/uploads/files/1502785891182-screen-shot-2017-08-15-at-4.30.28-pm.png](url)