pycom/pymakr-atom #33: serialport incorrect version

KoenR3, 2017-08-17:
1. install pymakr

**Atom**: 1.19.2 x64
**Electron**: 1.6.9
**OS**: Mac OS X 10.12.6
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.0.3

### Stack Trace

Failed to load the Pymakr package

```
At The module '/Users/koenrutten/.atom/packages/Pymakr/precompiles/serialport-osx/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '/Users/-----/.atom/packages/Pymakr/precompiles/serialport-osx/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/Pymakr/node_modules/bindings/bindings.js:76:44)
    at /packages/Pymakr/precompiles/serialport-osx/lib/bindings.js:3:35)
    at /packages/Pymakr/precompiles/serialport-osx/lib/bindings.js:36:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/Pymakr/precompiles/serialport-osx/lib/serialport.js:12:25)
    at /packages/Pymakr/precompiles/serialport-osx/lib/serialport.js:476:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/connections/pyserial.js:20:18)
    at /packages/pymakr/lib/connections/pyserial.js:26:30)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/board/pyboard.js:3:34)
    at /packages/pymakr/lib/board/pyboard.js:24:29)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at /packages/Pymakr/lib/main.js:7:21)
    at /packages/pymakr/lib/main.js:41:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app/src/package.js:887:35)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:158:34
    at Package.module.exports.Package.measure (/Applications/Atom.app/Contents/Resources/app/src/package.js:99:21)
    at Package.module.exports.Package.load (/Applications/Atom.app/Contents/Resources/app/src/package.js:144:18)
    at PackageManager.module.exports.PackageManager.loadAvailablePackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:565:22)
    at PackageManager.module.exports.PackageManager.loadPackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:495:27)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:643:36)
    at /Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:535:35
    at exit (/Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:17)
    at /Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:351:17
    at Socket.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:220:15)
    at emitOne (events.js:101:20)
    at Socket.emit (events.js:188:7)
    at Pipe._handle.close [as _onclose] (net.js:501:12)
```

### Commands

```
     -2:39.9.0 tree-view:show (atom-workspace.workspace.scrollbars-visible-when-scrolling)
     -0:49.9.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
advanced-open-file 0.16.6 
atom-beautify 0.30.5 
atom-dark-fusion-syntax 2.2.0 
atom-material-syntax 1.0.6 
atom-material-ui 2.0.4 
atom-mdtoc 0.8.3 
atom-ternjs 0.18.3 
atom-typescript 11.0.7 
atomatigit 1.5.5 
autocomplete-bibtex 0.7.0 
autocomplete-modules 1.6.10 
autocomplete-paths 2.8.0 
autocomplete-python 1.10.1 
autoupdate-packages 1.3.1 
blame 0.12.0 
busy 0.7.0 
busy-signal 1.4.3 
docblockr 0.11.0 
dockerletion 0.2.0 
easy-motion-redux 1.1.1 
ex-mode undefined 
expand-region 0.3.1 
file-icons 2.1.11 
fixmyjs 1.4.1 
git-history 3.3.0 
git-plus 7.9.3 
golden-ratio 0.3.0 
highlight-line 0.12.0 
highlight-selected 0.13.1 
intentions 1.1.5 
javascript-snippets 1.2.1 
json-path-finder 0.1.9 
language-diff 0.7.0 
language-docker 1.1.8 
language-latex 1.1.0 
language-terraform 0.8.1 
latex 0.45.0 
latexer 0.3.0 
lines 0.13.1 
linter undefined 
linter-docker 0.2.0 
linter-eslint 8.2.1 
linter-flake8 2.2.1 
linter-jshint 3.1.5 
linter-jsonlint 1.3.0 
linter-markdown 5.1.0 
linter-tslint 1.7.0 
linter-ui-default 1.6.7 
markdown-scroll-sync 2.1.2 
markdown-writer 2.6.5 
maximize-panes 0.2.0 
merge-conflicts 1.4.5 
minimap 4.29.6 
move-panes 0.2.0 
nucleus-dark-ui 0.12.3 
pdf-view 0.59.0 
platformio-ide-terminal 2.5.5 
project-plus 1.0.0 
proton-mode 0.16.0 
Pymakr 1.0.3 
python-isort 0.0.7 
python-tools 0.6.9 
python-yapf 0.14.0 
react 0.16.2 
react-snippets 0.9.0 
recent-files-fuzzy-finder 0.3.2 
relative-numbers 0.8.0 
release-notes 0.53.0 
tab-switcher 1.5.6 
theme-switch 0.4.0 
vim-mode-plus 0.97.2 
zentabs 0.8.8 
```

RalphHogenbirk, 2017-08-30:
Hi @KoenR3, please try the version on the develop branch at the moment and see if that fixes it. For info on how to install it, look at '[manual install](https://github.com/pycom/pymakr-atom#manual-install)' (download source, empty '~/.atom/packages/pymakr/` folder completely, put source there, run ```apm install``` from the same folder)

RalphHogenbirk, 2017-09-01:
Released today in 1.1.0. Let me know if it works for you.

KoenR3, 2017-09-02:
Yes this seems to work, I haven't been able to test with a pycom module yet though.  But I don't get any issues when starting up atom!

closed 2017-09-02 by KoenR3