pycom/pymakr-atom #34: Run program :

PhSurbay, 2017-08-22:
Since the last version of Atom and PyMakr, we can't run any program with the button run in Atom
We have alway this error :
```
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
NameError: name 'null' is not defined

```

I have remove and reinstall Atom, PyMaker, nodejs and npm 
But nothing work

In _apm -v_ we have : 
```
~$ apm -v
apm  1.18.4
npm  3.10.10
node 6.9.5 x64
python 2.7.13
git 2.11.0
```

Currently the sync work ..

So if you've got a solution

RalphHogenbirk, 2017-08-28:
I've got one more report of this, but I can't reproduce it on Ubuntu yet. Can you share your ```atom --version``` output?

PhSurbay, 2017-08-28:
```
~$ atom --version
Atom    : 1.19.2
Electron: 1.6.9
Chrome  : 56.0.2924.87
Node    : 7.4.0

```

For further information, I've two Lopy with two different firmware : '1.7.8.b1' and  '1.7.9.b3'

RalphHogenbirk, 2017-08-31:
Improved some code in the run feature and this seems to have solved this issue. It only occured if i opened a file outside of a project, not one inside a project. And only on ubuntu. 

Fix is on develop (commit bcc1c61) if you want to test it.

RalphHogenbirk, 2017-09-01:
Fix included in v1.1.0 that was just released

PhSurbay, 2017-09-04:
Great, that's seem good :)

closed 2017-09-04 by PhSurbay