pycom/pymakr-atom #35: Atom just updated and this message appeared.

thewindop, 2017-08-31:
Failed to activate the Pymakr package
The module '/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at bindings (/Users/andy/.atom/packages/Pymakr/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/lib/bindings.js:3:35)
    at Object.<anonymous> (/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/lib/bindings.js:36:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/lib/serialport.js:12:25)
    at Object.<anonymous> (/Users/andy/.atom/packages/Pymakr/precompiles/serialport-osx/lib/serialport.js:476:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/andy/.atom/packages/pymakr/lib/connections/pyserial.js:20:18)
    at Object.<anonymous> (/Users/andy/.atom/packages/pymakr/lib/connections/pyserial.js:26:30)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/andy/.atom/packages/pymakr/lib/board/pyboard.js:3:34)
    at Object.<anonymous> (/Users/andy/.atom/packages/pymakr/lib/board/pyboard.js:24:29)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/andy/.atom/packages/Pymakr/lib/main.js:7:21)
    at Object.<anonymous> (/Users/andy/.atom/packages/pymakr/lib/main.js:41:2)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (/Applications/Atom.app/Contents/Resources/app/static/<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (/Applications/Atom.app/Contents/Resources/app/src/package.js:887:35)
    at Package.module.exports.Package.activateNow (/Applications/Atom.app/Contents/Resources/app/src/package.js:242:22)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:225:38
    at Package.module.exports.Package.measure (/Applications/Atom.app/Contents/Resources/app/src/package.js:99:21)
    at /Applications/Atom.app/Contents/Resources/app/src/package.js:218:32
    at Package.module.exports.Package.activate (/Applications/Atom.app/Contents/Resources/app/src/package.js:215:40)
    at PackageManager.module.exports.PackageManager.activatePackage (/Applications/Atom.app/Contents/Resources/app/src/package-manager.js:645:40)
    at /Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:535:35
    at exit (/Applications/Atom.app/Contents/Resources/app/node_modules/settings-view/lib/package-manager.js:128:22)
    at triggerExitCallback (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:338:17)
    at ChildProcess.<anonymous> (/Applications/Atom.app/Contents/Resources/app/src/buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

RalphHogenbirk, 2017-09-01:
Very likely this is fixed in the just released v1.1.0 version of the plugin (only for 1.19.0 and higher). Let me know if that solves it.

caseyrodgers, 2017-09-01:
Nope, still busted with same error ...

RalphHogenbirk, 2017-09-01:
Exactly the same stack trace? Can you post it if it's any different, together with the output of your atom ---version?

thewindop, 2017-09-02:
ok so that works for me, however I am on 1.19.4 for ATOM. I see there is a .5 version in the pipeline.
I’m not going to update at the moment, if another release is needed?

rodgergr, 2017-09-02:
Hope this helps to track the problem down.

I have been having the same problems, running on WIN10 64 bit, so I

1. Uninstalled Pymakr and Atom
2. Installed Atom - shows as Version 1.19.5
3. Installed Pymakr - Version 1.1.0

Got following error and Stack Trace:-

"Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`
Hide Stack Trace
TypeError: "Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`
    at new SerialPort (C:\Users\Rodger\.atom\packages\Pymakr\precompiles\serialport-win\lib\serialport.js:119:11)
    at new PySerial (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/connections/pyserial.js:35:19)
    at file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/board/pyboard.js:199:28
    at Function.isSerialPort (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/connections/pyserial.js:125:7)
    at Pyboard.connect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/board/pyboard.js:196:14)
    at Pymakr.continueConnect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:246:21)
    at Pymakr.connect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:205:12)
    at new Pymakr (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:113:12)
    at Object.activate (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/main.js:17:19)
    at Package.module.exports.Package.activateNow (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:253:25)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:225:38
    at Package.module.exports.Package.measure (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:99:21)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:218:32
    at Package.module.exports.Package.activate (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:215:40)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package-manager.js:645:40)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\node_modules\settings-view\lib\package-manager.js:535:35
    at exit (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\node_modules\settings-view\lib\package-manager.js:128:22)
    at triggerExitCallback (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\buffered-process.js:338:17)
    at ChildProcess.<anonymous> (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

thewindop, 2017-09-02:
> at ChildProcess. (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\buffered-process.js:368:17)

I’m on mac

Darwin andys-MBP 15.4.0 Darwin Kernel Version 15.4.0: Fri Feb 26 22:08:05 PST 2016; root:xnu-3248.40.184~3/RELEASE_X86_64 x86_64

jadegraaf, 2017-09-04:
I had the same issue, after I reinstalled the Pymakr package it was resolved!

RalphHogenbirk, 2017-09-04:
@rodgergr Try the latest code on develop, that should solve the issue.

RonaldKrause, 2017-09-04:
@RalphHogenbirk :
Sorry cannot  confirm. Even after reinstalling Atom & pymakr (latest develop) the error persists: 
**Atom**: 1.19.5 x64
**Electron**: 1.6.9
**OS**: Win 10 pro
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.1.0
Failed to activate the Pymakr package
At "Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`

rodgergr, 2017-09-05:
Sorry @RalphHogenbirk 

As reported by @RonaldKrause, there has been no change.

I deleted existing Pymakr directories, downloaded the latest code on develop, installed with apm imstall, and restarted Atom, and the output looks identical to the previous stack trace, as below.

"Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`
Hide Stack Trace
TypeError: "Bindings" is invalid pass it as `options.binding` or set it on `SerialPort.Binding`
    at new SerialPort (C:\Users\Rodger\.atom\packages\Pymakr\precompiles\serialport-win\lib\serialport.js:119:11)
    at new PySerial (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/connections/pyserial.js:35:19)
    at file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/board/pyboard.js:199:28
    at Function.isSerialPort (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/connections/pyserial.js:125:7)
    at Pyboard.connect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/board/pyboard.js:196:14)
    at Pymakr.continueConnect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:246:21)
    at Pymakr.connect (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:205:12)
    at new Pymakr (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/pymakr.js:113:12)
    at Object.activate (file:///C:/Users/Rodger/.atom/packages/pymakr-atom-develop/lib/main.js:17:19)
    at Package.module.exports.Package.activateNow (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:253:25)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:225:38
    at Package.module.exports.Package.measure (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:99:21)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:218:32
    at Package.module.exports.Package.activate (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package.js:215:40)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package-manager.js:645:40)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package-manager.js:626:35
    at Config.module.exports.Config.transactAsync (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\config.js:346:24)
    at PackageManager.module.exports.PackageManager.activatePackages (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package-manager.js:621:25)
    at PackageManager.module.exports.PackageManager.activate (C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\package-manager.js:603:52)
    at C:\Users\Rodger\AppData\Local\atom\app-1.19.5\resources\app\src\atom-environment.js:843:36

RalphHogenbirk, 2017-09-05:
@RonaldKrause @rodgergr thanks for testing. After testing multiple new builds on my testmachine I get very mixed results. The serialport build is not including the bindings properly, but some combinations of versions seem to work, some times.... For now I've got a workaround for you that should fix it (commit a653d33 on the develop branch). I'll dive into this further.

RonaldKrause, 2017-09-05:
@RalphHogenbirk 👍  
Your patch works for me (with the main branch version). Sounds like a complicated thing to track down. Thank you for caring about the issue.

mdenhartog, 2017-09-05:
Downloaded the develop branch and installed the patch. Is working for me.

rodgergr, 2017-09-05:
It now works for me too:-) Thanks for your hard work.

RalphHogenbirk, 2017-09-06:
Good to hear. Tried today to get a better fix, but this workaround seems to work the best. The serialport lib simply builds to some different file formats and needs to link to a different js file of the lib.

Pushed it to master and published to Atom as v1.1.1

closed 2017-12-19 by RalphHogenbirk