pycom/pymakr-atom #36: Uncaught TypeError: No Binding set on `SerialPort.Binding`

humaninteractivedesign, 2017-09-01:
In an attempt to resolve the issue "Atom just updated and this message appeared - Failed to activate the Pymakr package #35" advise was to remove / replace with develop version of Pymakr and recompile npm install.  Pycom was attached during the r/r.  Pymakr threw this uncaught TypeError.

**Atom**: 1.19.5 x64
**Electron**: 1.6.9
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.0.3

### Stack Trace

Uncaught TypeError: No Binding set on `SerialPort.Binding`

```
At C:\Users\jeff_P50\.atom\packages\Pymakr\precompiles\serialport-win\lib\serialport.js:616

TypeError: No Binding set on `SerialPort.Binding`
    at Function.SerialPort.list (/packages/Pymakr/precompiles/serialport-win/lib/serialport.js:616:11)
    at Function.list (/packages/lib/connections/pyserial.js:138:16)
    at Pymakr.getSerial (/packages/lib/pymakr.js:155:14)
    at /packages/lib/pymakr.js:75:13)
    at emitNone (events.js:86:13)
    at PanelView.emit (events.js:185:7)
    at HTMLDivElement.PanelView.option_get_serial.onclick (/packages/lib/main/panel-view.js:143:13)
```

### Commands

```
```

### Non-Core Packages

```
build undefined 
keymaps undefined 
lib undefined 
menus undefined 
node_modules undefined 
precompiles undefined 
Pymakr 1.0.3 
spec undefined 
styles undefined 
```

simonjcarr, 2017-09-01:
Bump. Same issue for me. Does this mean the lopy devices we have purchased are unusable?

humaninteractivedesign, 2017-09-01:
Just received this-will ck if problems are resolved:

*Ralph* has posted a reply to: *Pymakr Atom Plugin released*:

V1.1.0 update. Includes a big refactor round that gets the code in line
with the VSCode plugin. This has very little effect on the UI and features,
except for a new config option 'open_on_start' that can be used to hide the
plugin by default. All problems with the serialport should also be resolved
(bindings file not found and serialport version mismatch errors), as long
as you are on the latest atom version (1.19 or higher)

   - Big refactor to get a shared base code with the VSCode plugin
   - Added 'open_on_start' config
   - Fixes issues with serialport lib for newer versions of atom
   - Bugfix in the run feature on linux.
   - More relaxed keep-alive protocol on telnet (ping has to fail twice
   before connection break)

Click here to read the full topic <https://forum.pycom.io/topic/1026>

Thanks!
*Pycom Forum*

*From:* Simon Carr [mailto:notifications@github.com]
*Sent:* Friday, September 01, 2017 10:33 AM
*To:* pycom/pymakr-atom <pymakr-atom@noreply.github.com>
*Cc:* humaninteractivedesign <jeffrey.gerth@humaninteractivedesign.com>;
Author <author@noreply.github.com>
*Subject:* Re: [pycom/pymakr-atom] Uncaught TypeError: No Binding set on
`SerialPort.Binding` (#36)

Bump. Same issue for me. Does this mean the lopy devices we have purchased
are unusable?

—
or mute
the thread

guyadams, 2017-09-01:
+1 for this - it seemed to be working under 1.0.3 but now I have upgraded, getting this same error - trying to downgrade back to 1.0.3 (under Atom 1.19.5) then gives me a node version incompatibility issue.

simonjcarr, 2017-09-01:
Why are so many people having so  many problems. There are so many problems and different versions of software that I am starting to feel overwhelmed and I am quite a technical person. I still can't get this to work. The PyCOM forum is useless because as a new user you have to wait for 10 minute between each post which prevents you from answer questions.

Who ever the team is behind this product need a leader to set some standards and some stability. I now have over £100 of hardware that is just junk without working software.

danicampora, 2017-09-01:
@simonjcarr your devices are not unusable. The plug-in was working perfectly fine on the previous version of Atom, but the last update broke it. Even without the plug-in you can do pretty much everything with the boards. Please wait until mid next week when we expect to have all the issues fixed. In the meantime, write your code using Atom and then connect to the WLAN network created by the LoPy, open FileZilla (or another FTP client) and connect to the FTP server, then transfer the files, reboot the module and your code will run. In order to access the REPL (the interactive Python prompt), any serial terminal (like putty) will work. All this is described in our documentation portal.

simonjcarr, 2017-09-01:
@danicampora Yep great, tomorrow I will got out and buy a WIFI card for my PC so I can use the boards, I wasn't expecting to have to buy more PC hardware to use the boards. You only have to do a quick google to see that these problems are not new, there have been reports almost every month over the last year of one thing or another not working due to incompatible versions of IDE's and plugin's.
When you look at create something, it's best to look at what others have done that are really successful like Arduino. They have a dedicated IDE and it works and they have control over the quality of it. It must be becoming obvious now that this experiment with IDE plugs is not working. The whole PyCom ethos seems to be around inspiring people, but when you can't rely on the software to get you over the first hurdle, people are probably going to find a new hobby or go back to Arduino with an IDE and software environment that just works. I am not saying that the developers are not working hard, I imagine they are overwhelmed with bug fixes, but I think the direction this project has taken is all wrong.

RalphHogenbirk, 2017-09-04:
I hear your feedback @simonjcarr and the last version of atom has been somewhat of an issue (for a small subset of users). On the other hand you see that the atom plugin has been downloaded about 5500 times and has been working well for so many people. Besides that, as Daniel is saying, the plug-ins are just an addition and can be used perfectly without any of these. Yes you need a wifi card in your pc or use a laptop, but that's a pretty fair assumption these days. Before the first pymakr was released, this was even the only option to use the wipy boards. 

Now what exact error are you getting with the v1.1.0 atom plugin? Can you post a dump of the stack-trace and a dump of your atom --version output?

aphulpin, 2017-09-04:
@RalphHogenbirk  Hi Ralph. Same issue, I uninstalled atom & pymakr, reinstalled it properly with dev & master plugin as well, nothing operational for now. Have you an idea of when you will propose a stable release regarding last atom changes  ? Regards, Alex

wwarodom, 2017-09-04:
I previously got atom v1.18 and it's working.  When I switch to atom 1.20.0-beta5 (on Windows7), I got exactly the same error stated above ("No binding set: Serial port").

simonjcarr, 2017-09-04:
Thanks for getting back to me. I understand you trying to defend the
plugin, but I still don't agree. I have seen the fix today on git hub for
this issue, I took a quick look at the code. The fix was change the path
where the plugin looks for files, to be specific to change the version
number of atom that is in the path.

In other words when Atom is upgraded again, which happens automatically
(The user has no choice), the plugin will be broken again. This is not a
solution that anyone would think was acceptable if this product is going to
have any sort of future.

For the time being I am going to continue testing using FTP and Telnet, but
I miss the stability and reliability of something like the Arduino echo
system and for that reason, I may well send my LoPy's back to RS Components
where I purchased them.

The only advice I can give you, is get one product right before you start
asking backers for more money for another product and as regards supporting
software, look at what is already hugely successful like Arduino and copy
it rather than trying to hack plugins together than just let the whole
thing down.

Good luck.

Simon

RalphHogenbirk, 2017-09-05:
@simonjcarr Yesterday's fix wasn't just the path. The whole serialport lib (most importantly the serialport.node) file was rebuild again. This happened to be done using 1.19.5 instead of 1.19.4 but they use the same node versions so it doesn't matter at all. From now on we'll be testing with the beta version as well, so when something breaks we'll be able to fix any issues with the new version before it's released.

I hear your feedback and I agree that it would be better if this had been fixed to be future proof from the start. But that's just not how it works with new products like this. The serialport lib turned out to be quite error-prone (because it has tons of dependencies on the OS and can't always be build on the users PC) and the solution we had for it from the start turned out to not be reliable enough when there were big updates in atom (for a subset of the users only). Sorry for the inconvenience you got out of it, but we are doing our best to improve the products as we go.

guyadams, 2017-09-05:
Do we have an ETA when the plugin on atom.io will be updated with the fix?

RalphHogenbirk, 2017-09-05:
@guyadams I just pushed a temporary fix on develop (see my last comment in [this topic](https://github.com/pycom/pymakr-atom/issues/35#issuecomment-327128632)). If more people confirm it works and I haven't found the permanent solution yet, I'll deploy a bugfix release somewhere tomorrow

aphulpin, 2017-09-05:
@RalphHogenbirk Thanks Ralph, works fine for me. Looking forward a stable version anyway ;-)

closed 2017-12-19 by RalphHogenbirk