pycom/pymakr-atom #37: "No file open to run" Linux Mint, Atom 1.19 latest dev

kuaka, 2017-09-01:
Running latest dev branch, in order to resolve the now closed [Failed to load the Pymakr package](https://github.com/pycom/pymakr-atom/issues/1#issuecomment-326173539) issue. 
I get this problem when trying to run a file in the console. The file is open. 
I couldn't get any file to run. Thinking that it may be related I applied the changes in the commit [bcc1c61](https://github.com/pycom/pymakr-atom/commit/bcc1c61b3bb4c3c635e819b5ec5981bce560607b)  refered to in the issue ["Run program : name 'null' is not defined #34"](https://github.com/pycom/pymakr-atom/issues/34)
It is still an issue, although the behaviour may have changed as I can now, sometimes get the file to run. 
If I open atom and click on the file in the project then hit run it usually works, if I then click on the file, say to make a change, then the run button results in the "No file open to run".

RalphHogenbirk, 2017-09-01:
Thanks for the feedback. issue #34 should have been solved in the latest commit (bcc1c61). I'll do some more testing on my linux vm now to see if I can reproduce. Have any idea what caused the change in behaviour that made it 'sometimes' work now?

RalphHogenbirk, 2017-09-01:
Pushed a small fix now that caused it to throw an error in some situations. Can you check if this also solves your problem?

kuaka, 2017-09-01:
Wow thanks for the super fast response. I have just tried it and it seems to have fixed it! fanatastic :)

closed 2017-09-01 by kuaka

satay99, 2018-04-27:
I seems to face the same problem all over again when I installed pymakr in Atom. This is my system setup.

OS : macOS version 10.12.6
Atom : 1.26.1 x64

The version of pymakr that I could install is version 1.0.3 as Atom do not allow me to install the latest pymakr update.

When I ran main.py. It threw the same message again.

Traceback (most recent call last):
File "<stdin>", line 2, in <module>
NameError: name 'null' is not defined.

Any help will be appreciated.

RalphHogenbirk, 2018-04-27:
@satay99 I see they deployed 1.26.1 today but the current Pymakr is only tested with 1.26 and lower. SO it reverts to a very old version back when there was no version limitation added. 
I'm planning a release either today or monday, to which support for 1.26.1 and 1.27.0 beta is added.

satay99, 2018-04-27:
Thank you RalphHogenbirk for the fast reply. I will wait for the new release. Thank you again.

satay99, 2018-04-29:
I installed 1.26.0 and Pymakr installed without any problem. However, when i tried to run the example show here -> https://docs.pycom.io/chapter/gettingstarted/programming/first_project.html, it throws an error as below

File "<stdin>", Line 7
SyntaxError: invalid syntax

Any idea what could it be ?

RalphHogenbirk, 2018-04-30:
@satay99 Seems like there is a typo in the example: It has 'While' with a capital W instead of a lowercase 'while'. If you run it like that, it works :) I'll ask someone to change the example.

satay99, 2018-05-01:
Thank you very much. That solve the problem.