pycom/pymakr-atom #38: Documentation does not reflect what I see in Atom

simonjcarr, 2017-09-01:
First of all I have an error with "Get Serial Ports". So I am not able to connect to the board. That issue has been reported https://github.com/pycom/pymakr-atom/issues/36 . Please can you fix that so I can use my board.

The following instruction is to navigate to 'Settings' > 'Global Settings'. However I don't see as settings menu. I do have a settings menu in the File menu, but there is no mention of 'Global Settings' in that menu. 

I am using Windows 10 Pro

I am starting to get worried that I have wasted a lot of money. Is there another IDE I can use?

RalphHogenbirk, 2017-09-01:
Hi, thanks for the feedback. The instruction 'settings > global settings' is referring to the 'Settings' button in the pycom console bar on the right bottom, above the terminal. Then it'll open the global settings of the plugin itself. You can also get there from your atom settings (ctrl + ,), packages, then press the 'settings' button for the Pymakr plugin.

Issue #36 is being looked into, though i think the error it describes is thrown on startup. If you have this error, you'll have more issues than just the 'get serial ports' feature :) Or is something else happening than in this other issue.

Can you share your versions so I can help you with it? (output of ```atom --version```)

In any case feel free to try the VSCode plugin as well if this one doesn't suit you. If you have issues with the plugins you can always use FTP for file uploads and another telnet or serial program like putty to use the REPL. But don't worry, we'll get you up and running.

simonjcarr, 2017-09-02:
Hi,

I have installed VSCode and installed the PyMakr extension as you suggested
as an alternative to Atom, however the instructions here
https://docs.pycom.io/chapter/pymakr/installation/vscode.html say

   1. Connect your Pycom device to your computer via USB. Remove the wire
   between GND and G23, before plugging in your device, if you have just
   finished upgrading your firmware!
   2. Open 'Global Settings' (this opens automatically after first install).

However Nothing opens after installing the extension and the settings cog
on the extension only gives you the option to disable or install the
extension. I don't see any configuration anywhere else for the PyMakr
extension.

I have managed to find a laptop. It has Windows and Wifi. By connecting to
the LoPy access point I was able to get a Telenet connection to 192.168.4.1
I could also connect on ftp to the same address.

I have two LoPy's I want to setup a LoPy to LoPy connection.

Some questions

1) When I plug both devices in they will both have a network address of
192.168.4.1. How do I work with that, how do I change them to connect to my
wifi router as clients. I assume I can't while I am connected to them over
wifi.
2) Assume that I just boot them up one at a time and put the sample code on
for LoPy to LoPy ( *https://docs.pycom.io/chapter/tutorials/lopy/lopy-lopy.html
<https://docs.pycom.io/chapter/tutorials/lopy/lopy-lopy.html> ) *using FTP,
do I have to use a specific file name or extension for the files that I
upload or does the board just read any file that is uploaded to it when it
is rebooted and start running the code. I can't see anything about this in
the instructions?
3) Is the PyMakr IDE still working? Are there any tutorials for it?
4) When will Atom be working again and will the PyMakr plugin break again
the next time Atom updates?

Thanks for your help
Simon

fluidzer0, 2017-09-12:
Hi Simon, 
#36 can be solved thus. I am also using Windows 10 pro.
1. After installing 'pymakr' plugin, REPL console opens on the right hand side bottom of the atom. It looks like this
[image]

2. To find which serial port your board has plugged in, 
    a. use the keyboard shortcut Win+X --> "Device Manager". See that your board must be listed in "ports(COM & LPT)". IF you click on it, you should be able to view your board with the port number as shown below.
[image]

   b. you could alternatively locate the serial port by following: in the REPL console in Atom
           More --> Get serial ports  (shown below)
 [image]

3. This serial port you need to enter in the "Device address" by going as- (In REPL console) 
   settings --> Global Settings --> 
  Scroll down and in Device Address, where default would be "192.168.4.1", replace it with your serial port number "COM#", (e.g. COM2 in my case). 
[image]

4. Come out of the settings and hit "connect on the REPL console".

5. You should be able to see the ">>>" prompt indicating that you can access the device.

That's it. Done! :)

P.S.- I some cases you may not be able to get the serial port i.e. may see a small yellow triangle in ports section, in that situation, try installing the FTDI drivers following the following the following steps: 
 1. Download the drivers:
[FTDI DRIVERS FOR WINDOWS ](http://www.ftdichip.com/FTDrivers.htm)
 2. WIN+X --> Device manager --> ports --> listed device --> right click --> update drivers -->Browse my computer for driver software --> select the downloaded drivers
3. installation complete.

Cheers.

closed 2017-12-19 by RalphHogenbirk