pycom/pymakr-atom #39: Can't run/sync via USB (/dev/ttyUSB0)

otmaster, 2017-09-01:
I hope this isnt dublicate...

I can use my device just fine via WIFI. I can connect and get this output:

`>>>
>>>
>>>`

 now i can execute my code, even sync. But when i try to use it via USB its getting difficult. The first attempt always fails. I have to push the reset-button. after that I get the following output:

`Disconnected. Click here to reconnect.
Connecting on /dev/ttyUSB0...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff9010,len:8
load:0x3fff9018,len:248
load:0x40078000,len:4056
load:0x4009fc00,len:920
entry 0x4009fde4
I (1633) wifi: frc2_timer_task_hdl:3ffd4ed0, prio:22, stack:2048
I (1649) wifi: Init lldesc rx mblock:10
I (1649) wifi: Init lldesc rx ampdu len mblock:7
I (1649) wifi: Init lldesc rx ampdu entry mblock:4
I (1651) wifi: pp_task_hdl : 3ffdc00c, prio:23, stack:8192
I (1656) wifi: sleep disable
I (2645) wifi: mode : softAP (24:0a:c4:00:5a:41)
MicroPython v1.8.6-470-gf41d0f27 on 2017-02-22; SiPy with ESP32
Type "help()" for more information.
>>>`

but i cant type anything or run or sync. just nothing happens.

Using Ubuntu 17.04

chendrix4, 2017-10-03:
Have the exact same issue on Windows (and inability to do anything on Linux). I actually used to be able to connect and sync code, and if I reconnected then Atom would lock up.

I had some small success restoring factory firmware w/ [safe boot mode](http://docs.micropython.org/en/v1.8.6/wipy/wipy/general.html#wipy-boot-modes) (note that for SiPy, you connect to P12 (20) and 3.3V).

Finally, I was able to connect more reliably in PuTTY or Python/IDLE using the serial module. Now, I cannot initiate any connection at all in any of these environments.

RalphHogenbirk, 2017-10-09:
Are both of you running the latest firmware on the boards as well? If you can't connect using PuTTY as well, there is something else going with the board itself. Make sure you have the following code on the top of your bootfile.

```import os
import machine
uart = machine.UART(0, 115200)
os.dupterm(uart)```

chendrix4, 2017-10-09:
I am. After erasing flash memory, I've been good so far and able to connect. My boot file does have that code at the top.

closed 2017-12-19 by RalphHogenbirk