pycom/pymakr-atom #40: Uncaught TypeError: Cannot read property '2' of undefined

DiegoAntonino, 2017-09-12:
[Enter steps to reproduce:]

**Atom**: 1.19.7 x64
**Electron**: 1.6.9
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [Pymakr](https://github.com/pycom/pymakr-atom) package 1.1.1

### Stack Trace

Uncaught TypeError: Cannot read property '2' of undefined

```
At C:\Users\<USER>\.atom\packages\Pymakr\node_modules\xterm\lib\SelectionManager.js:244

TypeError: Cannot read property '2' of undefined
    at SelectionManager._onSingleClick (/packages/Pymakr/node_modules/xterm/lib/SelectionManager.js:244:17)
    at SelectionManager._onMouseDown (/packages/Pymakr/node_modules/xterm/lib/SelectionManager.js:201:22)
    at /packages/Pymakr/node_modules/xterm/lib/SelectionManager.js:51:90)
```

### Commands

```
     -7:11.8.0 editor:newline (input.hidden-input)
     -7:00.7.0 core:cut (input.hidden-input)
     -6:58 core:copy (input.hidden-input)
     -6:56.8.0 core:paste (input.hidden-input)
  3x -6:55.1.0 core:undo (input.hidden-input)
     -6:52.6.0 core:cut (input.hidden-input)
     -6:48.6.0 core:paste (input.hidden-input)
  4x -6:45.7.0 core:delete (input.hidden-input)
  2x -6:34.1.0 core:save (input.hidden-input)
     -5:27 core:backspace (input.hidden-input)
     -5:20.1.0 core:save (input.hidden-input)
     -4:14 core:backspace (input.hidden-input)
     -4:13.5.0 editor:newline (input.hidden-input)
     -4:08.6.0 core:save (input.hidden-input)
     -4:01.7.0 core:delete (input.hidden-input)
     -3:59.9.0 core:save (input.hidden-input)
```

### Non-Core Packages

```
Pymakr 1.1.1 
```

RalphHogenbirk, 2017-12-19:
Is this still happening with the newer versions? Looks like a bug inside the xterm lib.

chrisvoo, 2018-02-01:
[Enter steps to reproduce:]

It just happened to me, but I wasn't able to understand why. I just unplugged and plugged in the device or something like that.

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.13.4
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.7

### Stack Trace

Uncaught TypeError: Cannot read property '2' of undefined

```
At /Users/ccastelli/.atom/packages/pymakr/node_modules/xterm/lib/SelectionManager.js:244

TypeError: Cannot read property '2' of undefined
    at SelectionManager._onSingleClick (/packages/pymakr/node_modules/xterm/lib/SelectionManager.js:244:17)
    at SelectionManager._onMouseDown (/packages/pymakr/node_modules/xterm/lib/SelectionManager.js:201:22)
    at /packages/pymakr/node_modules/xterm/lib/SelectionManager.js:51:90)
```

### Commands

```
```

### Non-Core Packages

```
atom-beautify 0.30.9 
atom-quokka 1.0.45 
atom-runner 2.7.1 
autocomplete-python 1.10.5 
busy-signal 1.4.3 
color-picker 2.2.5 
custom-title 1.0.1 
docblockr 0.13.6 
file-icons 2.1.16 
git-plus 7.10.0 
highlight-selected 0.13.1 
hyperclick 0.1.5 
intentions 1.1.5 
js-hyperclick 1.12.2 
language-babel 2.83.1 
language-pug 0.0.21 
linter 2.2.0 
linter-csslint 2.0.0 
linter-eslint 8.4.1 
linter-jshint 3.1.6 
linter-json-lint 0.1.2 
linter-markdown 5.2.0 
linter-pug-lint 0.2.0 
linter-pylama 0.9.5 
linter-ui-default 1.6.10 
node-debugger 1.10.1 
node-resolver2 2.0.2 
pymakr 1.2.7 
python-indent 1.1.4 
python-tools 0.6.9 
react 0.17.0 
```

RalphHogenbirk, 2018-02-05:
I'm going to update the xterm lib to the latest version (which seems to have some changes that I need to implement) and see if that solves the issues in the long run. Will keep you up to date.

romaas, 2019-04-12:
[Steps to reproduce ]

1.Install latest stable Atom 1.35.1 x64
2.Install Python 2.7.16
3.Install Pymakr 1.4.12

[Versions of software]

Operating system: Windows 10
Atom version: 1.35.1 x64
Pymakr Version: 1.4.12

I guess the issue isn't resolved yet.  I just got connected to the device ( I am using Expansion board v3.1 and FiPy) and printed "Hello World!" (I am just getting started with MicroPython and Pycom as well as Atom text editor) and then this error was thrown.

[image]

RalphHogenbirk, 2019-04-12:
@romaas no it hasn't. Newer version of xterm wasn't compatible with pymakr, so we reverted, but never actually took the effort to do a new integration of the latest xterm. Very rarely this error is still thrown, unfortunately. 
In Pymakr 2.0 which we're working on, will definitely (finally) fix it.

romaas, 2019-04-12:
@RalphHogenbirk  Thank you for letting me know. 
Hopefully it will be fixed in the next version.

johnsonthie, 2019-10-24:
I encountered this error after installing Pymakr 1.4.18 on Atom 1.41. OS: Windows 10 x64.

pedroknup, 2020-02-18:
That has been fixed with the new Pymakr 2.0

closed 2020-02-18 by pedroknup