pycom/pymakr-atom #42: Failed to load pymakr package

liowmei, 2017-09-22:
Hi, I'm new here. Currently facing this issue,
Error: The module '\\?\C:\Users\user\.atom\packages\Pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 49. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:66:33)
    at bindings (C:\Users\user\.atom\packages\Pymakr\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\Users\user\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings.js:3:35)
    at Object.<anonymous> (C:\Users\user\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings.js:36:3)
    at Module.get_Module._compile (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\user\.atom\packages\Pymakr\precompiles\serialport-win\lib\serialport.js:12:25)
    at Object.<anonymous> (C:\Users\user\.atom\packages\Pymakr\precompiles\serialport-win\lib\serialport.js:476:3)
    at Module.get_Module._compile (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/user/.atom/packages/pymakr/lib/connections/pyserial.js:20:18)
    at Object.<anonymous> (file:///C:/Users/user/.atom/packages/pymakr/lib/connections/pyserial.js:26:30)
    at Module.get_Module._compile (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/user/.atom/packages/pymakr/lib/board/pyboard.js:3:34)
    at Object.<anonymous> (file:///C:/Users/user/.atom/packages/pymakr/lib/board/pyboard.js:24:29)
    at Module.get_Module._compile (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\user\.atom\packages\Pymakr\lib\main.js:7:21)
    at Object.<anonymous> (file:///C:/Users/user/.atom/packages/pymakr/lib/main.js:41:2)
    at Module.get_Module._compile (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/user/AppData/Local/atom/app-1.20.1/resources/app/static/index.js:47:45)
    at require (internal/module.js:20:19)
    at customRequire (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\static\<embedded>:96:26)
    at Package.module.exports.Package.requireMainModule (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package.js:887:35)
    at C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package.js:158:34
    at Package.module.exports.Package.measure (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package.js:99:21)
    at Package.module.exports.Package.load (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package.js:144:18)
    at PackageManager.module.exports.PackageManager.loadAvailablePackage (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package-manager.js:565:22)
    at PackageManager.module.exports.PackageManager.loadPackage (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package-manager.js:495:27)
    at PackageManager.module.exports.PackageManager.activatePackage (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\package-manager.js:643:36)
    at C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\node_modules\settings-view\lib\package-manager.js:535:35
    at exit (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\node_modules\settings-view\lib\package-manager.js:128:22)
    at triggerExitCallback (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\buffered-process.js:338:17)
    at ChildProcess.<anonymous> (C:\Users\user\AppData\Local\atom\app-1.20.1\resources\app\src\buffered-process.js:368:17)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)

After reading the forum i found a solution which is to execute:

echo "python=/usr/bin/python2" >> ~/.atom/.apmrc

however, may i know how to actually execute this?

RalphHogenbirk, 2017-09-22:
That command is not the solution in this case. I think you might be on an old version of the plugin. Can you tell me which version of the plugin and which version of atom you are using?

liowmei, 2017-09-23:
Hi @RalphHogenbirk , for your information the atom version that i installed is as below : 
Product version : 1.20.1
File version : 1.20.1.0 

For plugin version : 
i have a doubt that when i choose the pymakr plugin beside the pymakr written as "pymakr 1.0.3" but after i installed it written as "pymakr 1.1.1"  Does this means the version of pymakr plugin? 

And may i know is it necessary to add a REPL console to Atom if yes then how to work with it? As it mentioned together while installing the plugin. "Adds a REPL console to Atom that connects to your Pycom board. It can run code on the board or synchronize your project files to it." 

Kindly thanks for your help. I trying to solve this problem but in vain.

liowmei, 2017-09-24:
Hi @RalphHogenbirk  , may i know how can i solve the issue?

venglebert, 2017-10-19:
Same problem on Windows 10 pro. Thanks

RalphHogenbirk, 2017-12-19:
This has been solved a while ago, finally closing it to clean up the list of issues :)

closed 2017-12-19 by RalphHogenbirk