pycom/pymakr-atom #43: Error compiling Less stylesheet

bjorn-bl, 2017-10-03:
I get this error message. It seems as there are paths inside the files that belong to Ralphs computer?

Atom is 1.20.1 x64

Error compiling Less stylesheet: C:\Users\Bjørn B. Larsen\.atom\packages\pymakr-atom-1.1.4\styles\pymakr.less
Line number: 7
'../node_modules/font-awesome/less/font-awesome.less' wasn't found. Tried - C:\Users\Bjørn B. Larsen\.atom\packages\pymakr-atom-1.1.4\node_modules\font-awesome\less\font-awesome.less,C:\Users\Bjørn B. Larsen\AppData\Local\atom\app-1.20.1\resources\app\node_modules\one-dark-syntax\node_modules\font-awesome\less\font-awesome.less,C:\Users\Bjørn B. Larsen\AppData\Local\atom\app-1.20.1\resources\app\node_modules\one-dark-ui\node_modules\font-awesome\less\font-awesome.less,C:\Users\Bjørn B. Larsen\AppData\Local\atom\app-1.20.1\resources\app\static\node_modules\font-awesome\less\font-awesome.less,C:\Users\Bjørn B. Larsen\AppData\Local\atom\app-1.20.1\resources\app\node_modules\font-awesome\less\font-awesome.less,..\node_modules\font-awesome\less\font-awesome.less

RalphHogenbirk, 2017-12-19:
Cleaning up old messages, see we didn't respond to this one. Is this error still happening? Looks like the package folder is a bit off. The 'pymakr-atom-1.1.4' folder should just be 'pymakr'. Did you do a manual install?

bjorn-bl, 2017-12-20:
Hi

The automatic installation failed, and I tried a manual one. It seems as the original problem with auto install was caused by the fact that my laptop was old. I have switched laptops but not yet installed atom on my new device.

/Bjørn BL.

Bjørn B. Larsen
Førsteamanuensis
NTNU, Inst. for elektroniske system

sdaniel55, 2018-07-06:
Hi @bjorn-bl, I'll close this issue for now. If your issue is still not resolved, feel free to reopen it. Thanks!

closed 2018-07-06 by sdaniel55