pycom/pymakr-atom #44: No syncing

HenkOosterhuis, 2017-10-04:
Communication with LoPy board works, even running files but sync produces error
Occurs in safeboot as well. FIles on LoPy are not rewritten.
Anyone any suggestions?

Firmware: 1.9.1.b1
Pymakr: 1.1.4
Atom: 1.21.0

>>>
>>>
Syncing project (main folder)...

Reading file status
Failed to read project status, synchronizing all files
Failed to read project status, synchronizing all files
[1/2] Writing file boot.py
[2/2] Writing file main.py
Synchronizing done, resetting board...
9489f0

{communication hangs}

RalphHogenbirk, 2017-10-05:
Strange, so you checked that the files are not written to the board using FTP? Looking at the log it only failed to read the project.pymakr json file from the board, but successfully writes the boot and main files Did this break after updating Pymakr or Atom, or did you try this for the first time now?

I just tried it on my windows machine and I seem to have another issue with the sync there (timeout for reading the file status). Will investigate, see if I can find any problems that also explain your issue.

RalphHogenbirk, 2017-10-05:
I just pushed a new release to the develop branch. This solves some issues that might interfere with the sync in some situations. But it doesn't entirely explain your report. 

If you want to try it out you can download the code from develop and install it manually (see: https://github.com/pycom/pymakr-atom/tree/develop#manual-install). Let me know if it solves your issue.

dritory, 2017-10-11:
The new release to the develop branch seems to solve the sync issues. I've not been able to sync with v.1.1.4 from the main branch and using Atom 1.20.1. I would urge to release this to the main branch as a hotfix, as syncing is not possible with the current version.

RalphHogenbirk, 2017-10-11:
Thanks for the feedback. I'm going to ask some more users to test with a few new features I put to the develop branch just now, together with the bugfix. If I get good feedback on that, I'll release all of this in a new version on Tuesday (I'll be out of office for a few days). I hope that works for you too. Feel free to use the develop version till that time.

closed 2017-10-11 by RalphHogenbirk