pycom/pymakr-atom #45: Error: Writing to COM port (GetOverlappedResult): Operation aborted)

jhoguaman, 2017-10-25:
I updated Atom (1.21.1) and Pymakr (1.2.0). The option Run in to REPL console works good. But when I want synchronize (Upload) a file, I get this error:
 
Error: Writing to COM port (GetOverlappedResult): Operation aborted
Failed to read project status, uploading all files

When I worked with previous versions  I didn't have any problem.

RalphHogenbirk, 2017-10-27:
Hi @jhoguaman, sorry to hear about this error. Looking at it, it should be a windows only issue. I didn't get it on my own windows machine when testing, but I'll try to reproduce it. Can you send me the versions you are using? (output of `apm --version`) 

I'm on a short trip outside the office until Monday, so I'll have a look at it first thing on tuesday. Hope that's not a problem for you. You could temporarily switch back to an older version for now, or use visual studio code

jhoguaman, 2017-10-27:
Hi @RalphHogenbirk, Do you refer to this?

[image]

Thanks for your response friend..!!

mkrummenacher, 2017-10-31:
I can confirm the problem on Windows 10 x64. Before the update it worked with expansion board and PySense. Now it's working if the LoPy is mounted on the PySense board, but not on the expansion board.

RalphHogenbirk, 2017-11-01:
Thanks for the extra info @mkrummenacher, that will help me narrow the problem down! I'll keep you guys up to date

danicampora, 2017-11-10:
@Smeedy thanks for the report. There is an issue with the latest firmware that is causing this. We are solving it and a new firmware release will be out within 48hrs.

ThomasBastien, 2017-11-13:
Is the new firmware available?

ThomasBastien, 2017-11-16:
The issue comes from Pymakr plugin for ATOM, with the release 1.1.5 it works.

agallo, 2017-11-20:
This is happening on Windows 7, Atom 1.22.1, Pymakr 1.2.1

mkrummenacher, 2017-11-21:
I have updated the firmware to 1.10.1.b1 but still have the same problem:
[image]

Windows 10 x64
Atom 1.22.1
Pymakr 1.2.1

Only uploading does not work, "show firmware version" and "run" are working.

shailes, 2017-11-22:
I also have precisely the same problem (and have been having it for the past 3 weeks or so, with different versions of Pymakr and the firmware):

Current configuration:
LoPy
Firmware 1.10.1.b1
Windows 10 x64
Atom 1.22.1
Pymakr 1.2.1

Upload doesn't work, but run does: 
> Failed to connect (Error: Writing to COM port (GetOverlappedResult): Operation aborted). Click here to try again.

pkilcoyne, 2017-11-24:
I'm seeing the same issue, uploads work fine from pysense board but on the expansion board I'm getting the above error. I'm on a windows 10 machine running Atom 1.22.1 and Pymakr 1.2.1

Has this been resolved yet or is there a work around?

Thanks,

Paul

RalphHogenbirk, 2017-12-07:
Pymakr v1.2.2 has now been released which should fix this bug! Let me know if it solves your issues

agallo, 2017-12-07:
Windows 7 64-bit, Atom 1.22.1, Pymakr v1.2.2 -- working so far. 

Thanks for the fix.

mkrummenacher, 2017-12-11:
I can confirm that it's working with the new Version. Windows 10 1709 x64, Atom 1.22.1, Pymakr 1.2.2.

Thank you for your help!

closed 2017-12-19 by RalphHogenbirk

SpoofEx, 2018-01-18:
It seems I  have a similar issue uploading a project:
[image]

when i remove the 036e59.....json file I get the following error:
[image]

Current configuration:
windows 10 1709 16299.125
atom 1.23.3
Pymakr 1.2.27

I noticed I had to limit the baud rate when I initially flashed the .bin file to the ESP32 otherwise it failed.