pycom/pymakr-atom #46: Pymakr Plugin 1.2.0 : Console didn't Toggle/Activate

Djsharma07, 2017-11-16:
Hello,

I have updated Pymakr Plugin for Atom to **1.2.0**, but it didn't show up. I've also tried to toggle it manually from Packages Menu in Atom. So it didn't activate. 

**_OS_**: (sysname='LoPy', nodename='LoPy', release='**1.10.1.b1**', version='v1.8.6-839-g536c958c on 2017-11-15', machine='LoPy withESP32', lorawan='1.0.0') 
**_Atom version: 1.21.2_**

Please solve the issue.

RalphHogenbirk, 2017-11-16:
Do you see any errors popping up in the developer console (View -> Developer -> Toggle Developer Tools)?
Can you tell us which OS you are running, as in Windows, mac, linux? A copy of the output of the ```apm --version``` command from your local terminal could also be helpful.

Djsharma07, 2017-11-16:
No. There's no error shown in the developer console. I'm using Windows 10.

RalphHogenbirk, 2017-11-16:
Ok, can you first try to uninstall the plugin completely and re-install it? I've seen an issue on Win10 once before where it had some trouble writing a few of the files when updating the plugin.

Djsharma07, 2017-11-16:
I've tried and this time it has shown error which is already given in issue #35 
What to do next?

Djsharma07, 2017-11-16:
I've tried again the same cycle and it toggled finally. I've tried uploading my code, it has shown a message and then got disconnected:

```
Uploading project (main folder)...

Reading file status
Failed to connect (Error: Writing to COM port (GetOverlappedResult): Operation aborted). Click here to try again.
Failed to read project status, uploading all files
Creating dir lib
```

rolandvs, 2017-11-18:
Updated (again ) to the latest update(s) of firmware 0.0.7 and latest release micropython. I cannot sync my files. Have atom 1.22.1 and tried to install pymakr 1.2.0, however forces me to use 1.0.3. So how to proceed?

RalphHogenbirk, 2017-11-18:
@rolandvs Ah, 1.22.1 is out, Pymakr plugin has compatibility set to <= 1.22.0, so that's why it refuses to install (and 1.0.3 is the last version where the compatibility field was not defined yet). I'll test if it works with the latest beta (v1.23.0) and will fix the compatibility field on monday.

RalphHogenbirk, 2017-11-20:
@rolandvs Should work now with v1.2.1. Let me know if it worked.

@Djsharma07 This GetOverlappedResult error is an issue on windows 64 specifically. We're working on the problem!

Djsharma07, 2017-11-20:
@RalphHogenbirk Okay! Do let me know as soon as it is solved!

rolandvs, 2017-11-20:
@RalphHogenbirk atom 1.22.1 noticed an update pymakr 1.2.1 and installed it. It seems to work now. BUT COPY PASTE FROM CONSOLE TO EDITOR IS STILL NOT WORKING.

RalphHogenbirk, 2017-12-07:
@Djsharma07 The GetOverlappedResult error is now resolved with v1.2.2. Let me know if it works!

@rolandvs I know, it's very inconvenient! I've tried solving it but didn't have enough time. At some point it just stopped working. Will look at this again soon. The paste is working again by the way, that was my fault, it's just the copy that doesn't work.