pycom/pymakr-atom #48: Atom config vs PyMakr config?

cparrish60, 2017-11-22:
Why is there a port setting for Atom and also a port setting for PyMakr config?  I'm trying to create separate projects for two different LoPys, but Atom treats them like one project with one fixed port number.  Also, what is the proper syntax to set the sync location?

RalphHogenbirk, 2017-12-19:
Hi @cparrish60, sorry for the delay, this message slipped through. 

First, which port setting are you referring to? The 'address' field? In the 'global settings' you can set the defaults, and with the 'Project config' you can set config per project. Make sure that you either have each project open in another Atom screen, OR if you have two projects open in the left navigation, the title of the project you are working on in the left navigation is selected. You can see which project is selected in the bar above the terminal.

The proper syntax for the sync location is: relative to the project folder. So if you have your files in a folder called ```lib```, which is inside your project, simply put ```lib``` in the config.

Does that answer your questions? Let me know if you need more help.

sdaniel55, 2018-07-09:
Hi @cparrish60, I'll close this issue since it looks like you've got the answer. If your problem is not resolved, feel free to reopen this. Thanks!

closed 2018-07-09 by sdaniel55

cparrish60, 2018-07-09:
It's resolved.  I used an Arduino.