pycom/pymakr-atom #49: Install failure atom 1.23.1

patmolloy, 2017-12-14:
Hi, The installer is saying it wants to install pymakr 1.0.3 ("the latest that is compatible with your version of Atom"). On installing I get ..

Installing “Pymakr@1.0.3” failed.Hide output…
fs.js:640
return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
^

Error: ENOENT: no such file or directory, open 'C:\Users\patmo.atom\packages\Pymakr\package.json'
at Error (native)
at Object.fs.openSync (fs.js:640:18)
at Proxy.fs.readFileSync (fs.js:508:33)
at Proxy.fs.readFileSync (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\asar-require\lib\require.js:86:27)
at C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\lib\install.js:476:42
at fn (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:582:34)
at Immediate.<anonymous> (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:498:34)
at runCallback (timers.js:649:20)
at tryOnImmediate (timers.js:622:5)
at processImmediate [as _immediateCallback] (timers.js:594:5)

kukaoo, 2017-12-14:
I have a similar issue.

RalphHogenbirk, 2017-12-14:
You can now update to Pymakr v1.2.3, which will solve the issue!

patmolloy, 2017-12-14:
Am still seeing this .. completely fresh install of atom.

Installing “Pymakr@1.2.3” failed.Hide output…
fs.js:640
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open 'C:\Users\patmo\.atom\packages\Pymakr\package.json'
    at Error (native)
    at Object.fs.openSync (fs.js:640:18)
    at Proxy.fs.readFileSync (fs.js:508:33)
    at Proxy.fs.readFileSync (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\asar-require\lib\require.js:86:27)
    at C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\lib\install.js:476:42
    at fn (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:582:34)
    at Immediate.<anonymous> (C:\Users\patmo\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:498:34)
    at runCallback (timers.js:649:20)
    at tryOnImmediate (timers.js:622:5)
    at processImmediate [as _immediateCallback] (timers.js:594:5)

RalphHogenbirk, 2017-12-14:
Sorry @patmolloy, I'll investigate this tomorrow and will post a solution for you!

RalphHogenbirk, 2017-12-18:
This issue has finally been fixed today. The problem was that Atom stopped supporting package names with uppercase. It's been changed to lowercase now. Unfortunately, you will lose the global package settings as a result of this.

closed 2017-12-19 by RalphHogenbirk