pycom/pymakr-atom #50: Pymakr 1.2.3 failed to install!

mjmillere80, 2017-12-15:
Everything was fine until I updated Pymakr today. Now no Pymakr so no REPL for Pycom. 

Installing “Pymakr@1.2.3” failed.Hide output…
fs.js:640
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^

Error: ENOENT: no such file or directory, open 'C:\Users\mjmil\.atom\packages\Pymakr\package.json'
    at Error (native)
    at Object.fs.openSync (fs.js:640:18)
    at Proxy.fs.readFileSync (fs.js:508:33)
    at Proxy.fs.readFileSync (C:\Users\mjmil\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\asar-require\lib\require.js:86:27)
    at C:\Users\mjmil\AppData\Local\atom\app-1.23.1\resources\app\apm\lib\install.js:476:42
    at fn (C:\Users\mjmil\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:582:34)
    at Immediate.<anonymous> (C:\Users\mjmil\AppData\Local\atom\app-1.23.1\resources\app\apm\node_modules\async\lib\async.js:498:34)
    at runCallback (timers.js:649:20)
    at tryOnImmediate (timers.js:622:5)
    at processImmediate [as _immediateCallback] (timers.js:594:5)

versamodule, 2017-12-15:
I have the same issue.

mjmillere80, 2017-12-15:
Found the exact same issue here:

https://forum.pycom.io/topic/2298/pymakr-installation-fail-on-atom-1-23-1-x64

I hope that the upcoming update will take care of this problem.

danicampora, 2017-12-16:
Hello guys,

Apologies for this issue. We are aware of it and a resolution is on the way. May I suggest trying Visual Studio Code in the meantime? Thanks.

Cheers,
Daniel

mjmillere80, 2017-12-16:
Any possibilities that version 1.2.2 could be made available until this is resolved?
Thx

RalphHogenbirk, 2017-12-18:
This issue has finally been fixed today. The problem was that Atom stopped supporting package names with uppercase. It's been changed to lowercase now in ~~Pymakr~~ pymakr version 1.2.5. Unfortunately, you will lose the global package settings as a result of this.

mjmillere80, 2017-12-18:
Thank you, I will check it out.

mjmillere80, 2017-12-18:
Back up and working just fine, Thank you

closed 2017-12-18 by mjmillere80