pycom/pymakr-atom #52: wrong path for bindings.

andrethemac, 2017-12-19:
after upgrading atom to 1.23.1 and pymakr 1.2.5. (using Linux Mint 18.3)
pymakr looks for bindings in the wrong path.
pymaker installs in directory ~/.atom/packages/pymakr (pymakr with an lowercase P) but bindings looks for  ~/.atom/packages/Pymakr/node_modules/bindings/bindings.js  (that is Pymakr with an uppercase P.
to get things working you need the pymakr twice, (pymakr  and Pymakr).

RalphHogenbirk, 2017-12-19:
Thanks for the report, I guess this is the result of the fact that 1.2.5 changes the package name to lowercase, where it was with an uppercase P before. I'll deploy a hotfix for it today!

andrethemac, 2017-12-19:
great :+1:

RalphHogenbirk, 2017-12-19:
@andrethemac It's hard to find where it decides to look in the Pymakr folder. There is no reference too it in the code. I'm getting the impression that Atom somehow remembers the old path with the uppercase P and tries to use it. Does the problem persist after restarting your pc and re-installing the Pymakr package? 

I also deployed a new version right now with some fixes that might help (v1.2.6). Let me know if it works.

andrethemac, 2017-12-19:
I've removed the config.cson as well, it contained an entry with Pymakr in it. rebooting and reinstalling did solve the problem. Thanks.

closed 2017-12-19 by RalphHogenbirk