pycom/pymakr-atom #53: Failed to load the pymakr package

maheshjan, 2017-12-19:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.5

### Stack Trace

Failed to load the pymakr package

```
At '../node_modules/font-awesome/less/font-awesome.less' wasn't found. Tried - C:\Users\Mahesh\.atom\packages\pymakr\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\one-dark-syntax\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\one-dark-ui\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\static\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\font-awesome\less\font-awesome.less,..\node_modules\font-awesome\less\font-awesome.less in C:\Users\Mahesh\.atom\packages\pymakr\styles\pymakr.less:7:0

LessError: '../node_modules/font-awesome/less/font-awesome.less' wasn't found. Tried - C:\Users\Mahesh\.atom\packages\pymakr\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\one-dark-syntax\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\one-dark-ui\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\static\node_modules\font-awesome\less\font-awesome.less,C:\Users\Mahesh\AppData\Local\atom\app-1.23.1\resources\app.asar\node_modules\font-awesome\less\font-awesome.less,..\node_modules\font-awesome\less\font-awesome.less
  at /packages/pymakr/styles/pymakr.less:7:0
```

### Commands

```
```

### Non-Core Packages

```
node-debugger undefined 
pymakr 1.2.5 
python-debugger 0.2.0 
```

RalphHogenbirk, 2017-12-19:
Thanks for the report. I just noticed this mistake as well. There is a reference error in there (to the 'Pymakr' folder with a capital, instead of lowercase). Will release it today in a hotfix.

RalphHogenbirk, 2017-12-19:
@maheshjan Released under v1.2.6. Let me know if it fixes your issue.

JamesMcCarthy79, 2017-12-20:
@RalphHogenbirk This release will not install:(

RalphHogenbirk, 2017-12-20:
@JamesMcCarthy79 can you tell me what error message you get when you try to install? What OS en atom version are you on?

maheshjan, 2017-12-20:
Its working for me. Thanks..

JamesMcCarthy79, 2017-12-20:
@RalphHogenbirk, Sorry it failed during update had to completely remove package and reinstall fresh and it worked fine. Thanks

RalphHogenbirk, 2017-12-20:
@JamesMcCarthy79 Ok good to hear! 
I noticed some issues as well that have to do with the fact we changed from Pymakr with uppercase to pymakr with lowercase. Atom seems to remember the old uppercase version somewhere in its internals, which can cause issues when updating.

maheshjan, 2017-12-20:
@RalphHogenbirk  Yes, you are right, I did the same.

hallard, 2017-12-21:
Atom: 1.23.1 x64
OS: Microsoft Windows 10 Pro
Thrown From: pymakr package 1.2.6

I tried uninstall/reinstall no way

```shell
Could not locate the bindings file. Tried:
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\build\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\build\Release\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\Debug\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\out\Release\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\Release\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\build\default\serialport.node
 → C:\Users\Charles-Henri\.atom\packages\Pymakr\node_modules\bindings\compiled\7.4.0\win32\x64\serialport.node
```

tried npm install serialport form `C:\Users\Charles-Henri\.atom\packages\Pymakr\` installed but same results

RalphHogenbirk, 2017-12-22:
Hi @hallard, your install also tries to get the bindings from the old 'Pymakr' folder instead of the lowercase 'pymakr'. I can't figure out why that happens, but it's done by Atom itself. Seems like it's more persistent on Windows. Try a PC reboot as well, while I try to figure out what's going on.

PeterMarkl, 2017-12-29:
Hi, I had the same error as @hallard but on 
OS: Ubuntu 16.04
Atom: 1.22.1
and pymakr: 1.2.6

Complete Uninstall, Reinstall, Reboot etc did not work for me.
I managed to solve the problem by deleting everything in this folder: ~/.config/Atom/Cache
After that, i Reinstalled the Pymakr package without errors.

corbo, 2018-02-10:
Still the same error here:
OS: Microsoft Windows 10 Pro
Atom: 1.23.3
pymakr 1.2.7

Tried
- an uninstall - re-install
- an uninstall, remove of the ./.atom/.amp/Pymakr and again a re-install

all with no success

RalphHogenbirk, 2018-02-12:
Hi @corbo, Which error do you have? The original 'font-awesome.less missing' or 'could not load bindings file'? Post a dump if you can.

corbo, 2018-02-12:
@RalphHogenbirk : 

`Could not locate the bindings file. Tried:
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\out\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\default\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\compiled\7.4.0\win32\x64\serialport.node
Show Stack Trace
The error was thrown from the pymakr package. This issue has already been reported.`

and the part with the stack trace:

`Hide Stack Trace
Error: Could not locate the bindings file. Tried:
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\Debug\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\out\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\Release\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\build\default\serialport.node
 → C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\compiled\7.4.0\win32\x64\serialport.node
    at bindings (C:\Users\Cor\.atom\packages\Pymakr\node_modules\bindings\bindings.js:88:9)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings\win32.js:2:36)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings\win32.js:101:3)
    at Module.get_Module._compile (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Cor/AppData/Local/atom/app-1.23.3/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:8:22)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\bindings\auto-detect.js:19:3)
    at Module.get_Module._compile (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Cor/AppData/Local/atom/app-1.23.3/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\index.js:9:17)
    at Object.<anonymous> (C:\Users\Cor\.atom\packages\Pymakr\precompiles\serialport-win\lib\index.js:17:3)
    at Module.get_Module._compile (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Cor/AppData/Local/atom/app-1.23.3/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Cor\AppData\Local\atom\app-1.23.3\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/Cor/.atom/packages/Pymakr/lib/connections/pyserial.js:23:7)
    at Object.<anonymous> (file:///C:/Users/Cor/.atom/packages/Pymakr/lib/connections/pyserial.js:34:30)`

If you would like to see anything else, please let me know.

RalphHogenbirk, 2018-02-12:
@corbo It's indeed the same problem, it tries to look in 'Pymakr' instead of 'pymakr' (lowercase). One more thing to try is the suggestion by @PeterMarkl above: removing ~/.config/Atom/Cache (and likely uninstalling and installing the pymakr package again). Let me know if that helps

corbo, 2018-02-14:
If it is looking in the folder Pymakr why not rename the pymakr folder to Pymakr?
Tried that and bingo! that works. 
It might present a  problem in the future updating the package but for now ik love this simple sollution.
I do remember, from a long time back, in using (C) libraries that there are some differences in case sensitivity. As this issue with updating is kinda a surprise/puzzle, that (libraries) would be a path that I would include in the search.

RalphHogenbirk, 2018-02-14:
@corbo haha if it works, it works :) But indeed it will cause problems on the next update. A few versions ago Pymakr was with a capital, but Atom stopped supporting capital letters so we had to go to lower case. Yet somewhere in Atom itself it has some cache/config that is still remembering the old uppercase version, which is causing this error.

mkelley88, 2018-02-25:
I'm getting the same error: "Failed to load the pymakr package". Atom told me this issue had already been reported and sent me here. Any idea how to get this working? 

Steps to reproduce:
1. I installed Pymakr with Atom' s built-in package manager
2. This error appears every time I start Atom.

**Atom**: 1.24.0 x64
**Electron**: 1.6.16
**OS**: Ubuntu 17.10
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.7

### Stack Trace

Failed to load the pymakr package

```
At Cannot find module './utils/CircularList'

Error: Cannot find module './utils/CircularList'
    at Module._resolveFilename (module.js:470:15)
    at Module._resolveFilename (/usr/share/atom/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (/usr/share/atom/resources/app/src/module-cache.js:354:58)
    at Module.require (/app.asar/static/index.js:40:43)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/node_modules/xterm/lib/Buffer.js:3:22)
    at /packages/pymakr/node_modules/xterm/lib/Buffer.js:98:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/node_modules/xterm/lib/BufferSet.js:13:16)
    at /packages/pymakr/node_modules/xterm/lib/BufferSet.js:67:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (/usr/share/atom/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/node_modules/xterm/lib/xterm.js:3:19)
    at /packages/pymakr/node_modules/xterm/lib/xterm.js:1580:3)
    at Module.get_Module._compile (/usr/share/atom/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/usr/share/atom/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.2.7 
```

Moniekaj, 2018-03-16:
I've reinstalled Atom and Pymakr. Version 1.2.9 (Pymakr) manually installed, Atom only installs version 1.0.3 (Pymakr).
At version 1.0.3, the difference in error is "NODE_MODULE_VERSION 49."

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.25.0 x64
**Electron**: 1.7.11
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.9

### Stack Trace

Failed to load the pymakr package

```
At The module '\\?\C:\Users\user\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\user\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:76:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:101:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.25.0/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.25.0/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:9:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.2.9 
```

RalphHogenbirk, 2018-03-16:
@Moniekaj Pymakr is not compatible with Atom 1.25 yet, but we're fixing that today. We need to re-compile the serialport lib that ships with it, because the nodejs version of atom has changed in this version.

@mkelley88 your error report looks like something coming from the xterm lib. Never seen it before, but glad it's solved by just re-installing. I'll try to update my ubuntu test machine to 1.7 so I can help debug a bit better in the future.

RalphHogenbirk, 2018-03-17:
@Moniekaj Just a heads-up, we released pymakr v1.2.10 which should work with atom 1.25, at least on OSX and Win64.

hase-berlin, 2018-03-18:
Well, it kinda-sorta works with Linux as well. But not always.
On my workstation, I get
The module '/home/hase/.atom/packages/pymakr/precompiles/serialport-linux/build/Release/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

but I get not error (and pymakr is working) on my Laptop.
Strangly, both are running Debian9(stretch).

Seems like an easy-to-solve version conflict, but I have no clue around npm and this Javascript/node.js stuff.

sdaniel55, 2018-07-09:
Looks like this issue has been resolved. If you're still experiencing problems, feel free to reopen it and provide some details. Thanks!

closed 2018-07-09 by sdaniel55