pycom/pymakr-atom #54: Uncaught Error: watch ~/pymakr.conf ENOSPC

aryaabhi436, 2017-12-22:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Ubuntu 16.04.3
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.6

### Stack Trace

Uncaught Error: watch /home/abhiarya/pymakr.conf ENOSPC

```
At fs.js:1318

Error: watch /home/abhiarya/pymakr.conf ENOSPC
    at exports._errnoException (util.js:1022:11)
    at FSWatcher.start (fs.js:1316:19)
    at Object.fs.watch (fs.js:1341:11)
    at /packages/pymakr/lib/main/settings-wrapper.js:43:33
    at FSReqWrap.oncomplete (fs.js:112:15)
```

### Commands

```
     -2:38.2.0 editor:newline (input.hidden-input)
     -2:34.3.0 core:paste (input.hidden-input)
 17x -2:25.9.0 core:move-left (input.hidden-input)
     -2:23.2.0 core:backspace (input.hidden-input)
     -2:20.1.0 core:move-down (input.hidden-input)
     -2:17.9.0 core:move-up (input.hidden-input)
  2x -2:11.7.0 core:backspace (input.hidden-input)
     -2:09.6.0 autocomplete-plus:confirm (input.hidden-input)
     -1:59 core:backspace (input.hidden-input)
     -1:38.2.0 editor:newline (input.hidden-input)
     -1:16.6.0 core:backspace (input.hidden-input)
     -0:06.3.0 tree-view:remove-project-folder (span.name.icon.icon-file-directory)
```

### Non-Core Packages

```
atom-clock 0.1.15 
busy 0.7.0 
console-panel 0.3.0 
microbit-micropython 0.3.2 
proto-repl 1.4.22 
pymakr 1.2.6 
termination 0.6.1 
tool-bar 1.1.0 
```

RalphHogenbirk, 2017-12-22:
HI @aryaabhi436 ENOSPC means there is no space left on the drive. Could that be the problem on your harddisk?

sdaniel55, 2018-07-09:
Hi @aryaabhi436, I'll close this issue due to inactivity. If your problem is not resolved, feel free to reopen this. Thanks!

closed 2018-07-09 by sdaniel55