pycom/pymakr-atom #55: Failed to activate the pymakr package

sandman, 2017-12-28:
[Enter steps to reproduce:]

1. Installed pymakr from Atom. Closed it.
2. Connected FiPy+PyTrack and started Atom.

**Atom**: 1.23.1 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 7 Home Premium 
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.6

### Stack Trace

Failed to activate the pymakr package

```
At Cannot find module './parsers'

Error: Cannot find module './parsers'
    at Module._resolveFilename (module.js:470:15)
    at Module._resolveFilename (~/AppData/Local/atom/app-1.23.1/resources/electron.asar/common/reset-search-paths.js:35:12)
    at Function.get_Module._resolveFilename (~/AppData/Local/atom/app-1.23.1/resources/app/src/module-cache.js:354:58)
    at Module.require (/app.asar/static/index.js:40:43)
    at require (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:10:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.23.1/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/connections/pyserial.js:23:7)
    at /packages/pymakr/lib/connections/pyserial.js:34:30)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.23.1/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/board/pyboard.js:3:34)
    at /packages/pymakr/lib/board/pyboard.js:22:29)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.23.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.23.1/resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
```

### Commands

```
```

### Non-Core Packages

```
pymakr 1.2.6 
```

RalphHogenbirk, 2017-12-28:
Hi @sandman, thanks for reporting. Can you check if your ```/packages/pymakr/precompiles/serialport-win/lib/``` folder contains another folder called 'parsers', which has 6 files in it including an index.js? I know of some cases on windows where some of the files in the precompiles folder don't get updated properly, maybe that's the case here. Uninstalling the plugin and re-installing it is worth trying.

sandman, 2017-12-28:
Hi @RalphHogenbirk, yes the parsers folder has 6 files including an index.js. Shall I uninstall/reinstall in any case?

RalphHogenbirk, 2017-12-28:
Ok, yes try that anyway @sandman

sandman, 2017-12-28:
Reinstalling the package worked. I guess this issue can be closed? 
Thanks @RalphHogenbirk

RalphHogenbirk, 2017-12-28:
Good to hear @sandman. I have this issue on my list to figure out why this sometimes happens. But in the meantime, I'll close it

closed 2017-12-28 by RalphHogenbirk

steek1337, 2018-03-16:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.25.0 x64
**Electron**: 1.7.11
**OS**: Unknown Windows version
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.9

### Stack Trace

Failed to activate the pymakr package

```
At The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).

Error: The module '\\?\C:\Users\Stefan\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 53. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at bindings (/packages/pymakr/node_modules/bindings/bindings.js:76:44)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:2:36)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/win32.js:101:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.25.0/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:8:22)
    at /packages/pymakr/precompiles/serialport-win/lib/bindings/auto-detect.js:19:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.25.0/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:9:17)
    at /packages/pymakr/precompiles/serialport-win/lib/index.js:17:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.25.0/resources/app/src/native-compile-cache.js:106:36)
```

### Commands

```
```

### Non-Core Packages

```
atom-terminal-panel 4.4.4 
gcc-make-run 0.2.12 
minimap 4.29.8 
pymakr 1.2.9 
```

RalphHogenbirk, 2018-03-16:
Hi @steek1337,  Thanks for reporting. I noticed yesterday that the 1.25.0 beta indeed doesn't work with Pymakr yet because of a change in nodejs version, causing the precompiled serialport lib to fail. Now I see they released it, so I'll hurry and build a new version that will work with 1.25. Hang in there for a Pymakr update. Downgrade to atom 1.24.0 to keep using Pymakr for now.

steek1337, 2018-03-16:
@RalphHogenbirk Okay,... I have had alot of problems with getting pymakr to work.... its always something:S

steek1337, 2018-03-16:
@RalphHogenbirk  how do I Downgrade atom 1.24.0?