pycom/pymakr-atom #56: Atom does not connect to Pycom board

galiniEdi, 2017-12-28:
Hello,

I am using Atom 1.22.1 and a GPy with an Expansion board v2.0. I am trying to connect the GPy via USB following https://docs.pycom.io/chapter/pymakr/installation/atom.html. I updated the device address to reflect the USB address but Atom attempts to connect to the board indefinitely, without succeeding. 
When I click the "Connect" button I get the message "Failed to connect (Error: Port is not open). Click here to try again."
I have tried to connect via command line with the screen command and it works fine, but Atom fails. 
Any idea what is wrong?

Thank you

RalphHogenbirk, 2017-12-28:
If it connects via command line with 'screen', it should indeed also work with Pymakr. Make sure you properly close the commandline connection before you use Pymakr and have no other connections open. Double check if you set it up to the exactly correct comport (use 'more -> list serialports, it automatically copies the top option to the clipboard, paste that in the global or project settings). Also make sure you are on the latest firmware for the GPy, and maybe even update to atom 1.23.1, just in case that matters (and let me know if it did)

If that doesn't help let me know and I'll try to debug it with you.

galiniEdi, 2017-12-29:
I have done everything you suggested apart from the firmware update. I am trying to update the firmware, following the instructions on the website. I have disconnected the device, used a jumper cable between G23 and GND, then connected the device and run the update. However neither the TUI or GUI forms work as they both say they cannot connect to the device. The GUI form also said that it "can't figure out if this is a Pysense/Pytrack".

galiniEdi, 2017-12-29:
Also (if this is of any help) when I am running the GUI for the update (with "python2 lopyupdate.py"), in the "Pycom Upgrade" page I get this errors when I click next:

TypeError: invalid result from CustomWizardPage.nextId(), an integer is required
TypeError: invalid result from CustomWizardPage.nextId(), an integer is required
TypeError: invalid result from CustomWizardPage.nextId(), an integer is required
QWizard::next: Page 0 already met

Xykon, 2017-12-29:
First off please make sure that you have closed Atom and/or Visual Studio Code and/or any other program that connects to the serial port. Once you connect the board with the jumper cable attached between GND + G23 press the reset button on the GPy. Then start the firmware update tool.

If it still doesn't work then please check with screen if when pressing the reset button on the GPy with the jumper cable attaches, you see "Waiting for download" as the last message. This message indicates that the module is in firmware update mode. If you don't see the message but instead you get the MicroPython prompt then the firmware update wire isn't set properly.

galiniEdi, 2017-12-29:
Thank you for the help. Pressing the reset button worked indeed and now the GPy is updated. Howver the initial problem remains as I cannot connect it to Atom. I have no other connections open (command line or Visual Studio or anything else) and I follow the instructions of https://docs.pycom.io/chapter/pymakr/installation/atom.html. The device address is that of the USB and when I click connect Atom attempts to connect indefinitely without success.

RalphHogenbirk, 2017-12-29:
Sorry to hear that @galiniEdi. Can you send me a screenshot of your terminal after one or two of those failing connection attempts? And can you try to connect to your board over wifi to make sure it's narrowed down to an issue with the serialport?

galiniEdi, 2017-12-29:
Attached is a screen-shot of the problem, along with the settings. However I don't think it is an issue of the serial port as I have been able to connect and run sample code using "screen".

[image]

RalphHogenbirk, 2017-12-29:
Thanks, got it. Try the wifi part anyway, then we know if it's the serialport part in Pymakr, or a more general Pymakr issue. Or even a problem with the micropython REPL for example.

From your screenshot i'm assuming you are using linux, should have asked that to begin with :) I'll do a test with these versions on linux to see if I can reproduce the issue.

galiniEdi, 2017-12-29:
Yes, sorry for not mentioning. I am using Ubuntu 16.04. I am trying the WiFi option at the moment.

Xykon, 2017-12-29:
Just wondering if you've used the same user account (without using sudo) to do the firmware update / screen access and to run VS Code. If not you might need to give your user access to the dialout group. Run **sudo usermod -a -G dialout $USER** and logout and/or restart.
Also can you please check with lsof /dev/ttyUSB0 that no other process is trying to access the serial port?

galiniEdi, 2017-12-29:
@Xykon thank you very much. Seems like your usermod suggestion did the trick. I can now connect and run code. Thank you very much for all the help.

closed 2017-12-29 by RalphHogenbirk

galiniEdi, 2017-12-29:
@RalphHogenbirk I managed to make it work, however just to answer your question, it did work with WiFi even before the solution. Thank you very much as well for all the help!

RalphHogenbirk, 2017-12-29:
Thanks @galiniEdi, good to hear! Credits to @Xykon!