pycom/pymakr-atom #57: Excessive CPU used when PyMakr console maximised

kompleteb, 2018-01-07:
I have recently converted across to the Atom Pymakr plugin and found an issue with excessive CPU being consumed by the plugin when the console window is maximised (i.e. the corporal stripe is pointing down). Here is a screenshot of my task manager under Windows 10 64-bit when the issue is seen.

[image]

The CPU is always above 30% (on an Intel i7 3.2Ghz)). The state of Atom with pymakr console maximised is pictured below:

[image]

However when the window is minimised....

[image]

the CPU usage is as expected...

[image]

Any ideas as to the cause? Everything works fine - its just really inefficient and my CPU fan drives me nuts after a while :)

RalphHogenbirk, 2018-01-07:
Hey @kompleteb, that's strange, never noticed that, will test it when I get my hands on a windows machine. On mac the atom cpu usage stays unchanged with the console open or closed. Can't think of any reason why this would happen, at least not on nodejs level.

Does the CPU also increase when connecting over IP?

kompleteb, 2018-01-07:
@RalphHogenbirk, good question...and the answer is:

CPU usage is normal when connecting over IP.  

Thanks for the tip.  Perhaps the serial routines in Nodejs are polling the input buffer too aggressively?

kompleteb, 2018-01-07:
and the peace and quiet of a silent fan is bliss!

RalphHogenbirk, 2018-01-07:
That's good to know, a serial routine polling too often makes sense. I see that the implementation of the serialport lib for windows has a lot of differences with the osx version, so the issue could be in there. Will put it on my list to look at. Curious to see if there are more people with the issue as well.

closed 2020-02-18 by pedroknup