pycom/pymakr-atom #58: Upload Failed

semireg, 2018-01-09:
I have an almost 100% reproducible issue with "upload failed." This is making working on our project nearly unusable. Please let me know how I can begin troubleshooting.

pymakr version 1.2.7:
```
Reading file status
Failed to read project status, uploading all files
Creating dir cert
[1/13] Writing file xyz.py
Upload failed. Please reboot your device manually.
```

RalphHogenbirk, 2018-01-09:
Hi @semireg, most common cause is an outdated firmware. Which version are you on? Try the latest version if you're not on it yet

semireg, 2018-01-09:
Hi @RalphHogenbirk, thanks for the reply. I'm building my firmware from source - latest master.

`(sysname='WiPy', nodename='WiPy', release='1.12.0.b1', version='96a2f3a-dirty on 2018-01-08', machine='WiPy with ESP32')`

semireg, 2018-01-09:
The issue seems to not be deterministic. There isn't much pattern... however, I can tell you that if I try to sync before main.py is called (before "Starting App" is shown) the whole thing gets mucked up and I must unplug/replug and press the reset button multiple times to get it working again.

```
Upload done, resetting board...
No files to upload
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:1060
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:15072
entry 0x400a063c
Starting App
```

^ I have to see my output "Starting App" before I try syncing again.

semireg, 2018-01-10:
Still suffering from this. There seems to be a 10% chance that at least one file will fail the upload. Now that I have 18 source files the chances are very high that at least one file will result in a failure. When this happens, it seems `project.pymakr` is either not written or corrupted. This means that all files need to be uploaded on the next round - greatly exacerbating this issue and causing high levels of frustration.

One idea would be to upload one file at a time, updating `project.pymakr` each time. Additionally, since we don't want `project.pymakr` to become corrupted on upload, each file could be uploaded to `.filename` - if successful, `filename` would be removed and `.filename` would be copied into its place.

danicampora, 2018-01-10:
@semireg we have a fix for this. An update to Pymakr with the patch and a new firmware release will be out next week.

danicampora, 2018-01-10:
A workaround in the meantime is to do a safe boot before syncing.

semireg, 2018-01-10:
Is this related to the watchdog timer being triggered?
Do you need a beta tester? Can I build from source?

semireg, 2018-01-10:
Safe boot does not remedy the issue. I connect G28 (P12) to 3V3 via pushbutton, hold the button, and press reset. While holding the safe button, the amber light blinks slowly, then faster, then fastest. I release the safe button. I get a REPL and try uploading. Still fails.

Ref - https://docs.pycom.io/chapter/toolsandfeatures/bootmodes.html

semireg, 2018-01-10:
The pymakr debugging and logging is in a pretty bad state. Certain errors aren't percolated up through the callbacks.

The error "Upload Failed" is not helpful, and worse, it's generated dynamically by gluing together variables - it makes it more difficult to debug.

If I add some debugging to __start_sync I see my issue is always: "Failed to confirm file transfer"

I found this after adding the console.log in this snippet:

```  __start_sync(oncomplete,method){
    this.logger.info("Start sync method "+method)
    var _this = this

    var cb = function(err){
      if(err) {
        console.log("Sync finished w/ error 1: " + err)
      }
      _this.sync_done(err)
    }
```

Then, sync_done tries to display an error... but it fails because err is a string, not an error object.

```
    if(err){
      mssg = this.method_name+" failed."
      mssg += err.message && err.message != "" ? ": "+err.message : ""
      if(this.in_raw_mode){
        mssg += " Please reboot your device manually."
      }
```

As an aside, `getErrorMessage` on line 477 of pyboard.js returns an empty string if the error can't be "found" in "error_messages" ... which seems suspicious.

The actual error "Failed to confirm file transfer" is generated in monitor.js:

```
  requestAck(cb){
    this.pyboard.send_cmd_read('\x00\x00',3,function(err){
      if(err){
        err = "Failed to confirm file transfer"
      }
      cb(err)
    },3000)
  }
```

"Silly" log at https://gist.github.com/semireg/d0527ec52fe6debcb5fe942fc1607e2e

semireg, 2018-01-11:
In the interim I'm having 100% success by copying single files using ampy. Here's a quick script I wrote that maintains shadow copies (hidden dot files) of files that reside on the WiPy. It simply uses diff to check for changes.

```
#!/bin/bash
USB_SERIAL=$(ls /dev/tty.usb*)
BAUD=115200
AMPY_CMD="ampy -p $USB_SERIAL -b $BAUD"
REM_DEST="/flash"

if [ "$1" == "-l" ]; then
  $AMPY_CMD ls $REM_DEST
  exit
elif [ "$1" == "-r" ]; then
  $AMPY_CMD rm $REM_DEST/$2
  exit
fi

COPIED=0

for FILE in "$@"; do
  if [ ! -f $FILE ]; then
    echo "$FILE not found"
    continue
  fi

  FILE_BASE=$(basename $FILE)
  FILE_DIR=$(dirname $FILE)

  TMP_FILE="$FILE_DIR/.$FILE_BASE.put"
  diff $FILE $TMP_FILE 2>&- > /dev/null
  STATUS=$?

  if [ $STATUS -ne 0 ]; then
    echo -n "Copying $FILE ... "
    $AMPY_CMD put $FILE $REM_DEST/$FILE_BASE
    if [ $? == 0 ]; then
      echo "Done"
      cp $FILE $TMP_FILE
      COPIED=1
    fi
  fi
done

if [ $COPIED -gt 0 ]; then
  echo "Resetting and connecting..."
  $AMPY_CMD reset
  screen $USB_SERIAL $BAUD
fi
```

Usage:
```
$ bin/ampy-put.sh src/*.py
Copying src/helpers.py ... 
```

RalphHogenbirk, 2018-01-17:
Hi @semireg, thanks for all the info and debugging you did in this thread. Sorry for my late response, i've been busy at CES las vegas and now I'm sick so not making many hours. For now some short comments on some messages above:

First about the code you're debugging in the plugin:
- The requestAck method should indeed not make an error string, but an error object. If this is fixed it'll display the proper error.
- The 'getErrorMessage' function is used to detect if a known error keyword is found inside any of the output of the board, and if so returns some pre-defined error messages. This is a separate feature from the normal error system using the err variable in the callback params, so it doesn't interfere with other errors when it returns an empty string.

Anyway, the problem remains to be somewhere in the file transfer protocol itself (causing the 'ack' to fail after sending one of the chuncks of a file). it only happes in some cases, which makes me think it has to do with the file contents. Or, as another users log made me suspect, it happens when it encounters a file that's in a dir 2 levels deep. I'll definitely do some tests to try to reproduce the issue.

We've been debugging and improving this home-made protocol to make it more and more stable, but problems like this keep turning up. We're planning to overhaul this whole protocol and replace it with something solid like webrepl or rshell. We'll be brainstorming that soon.

 If you have any ideas or want to help out, you're more than welcome to!

RalphHogenbirk, 2018-01-24:
Hi @semireg, so I found and fixed one bug in the upload feature. There was an issue with an escape character in the upload protocol, which ment that when the length of the filename (path+file) is exactly 27 characters long, it would fail. Looking at your logs it's possible that it affects you as well.  The fix is on the develop branch right now so you can test it if you want. It'll be included in a release later this week.

sdaniel55, 2018-07-09:
Hi @semireg, I think this issue has been resolved, so I'll close this. If you're still having problems, feel free to reopen it. Thanks!

closed 2018-07-09 by sdaniel55