pycom/pymakr-atom #60: Uncaught TypeError: Cannot read property 'length' of undefined

frocha27, 2018-01-15:
[Enter steps to reproduce:]
just try to copy the console text using Ctrl-C

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Microsoft Windows 7 Professional 
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.7

### Stack Trace

Uncaught TypeError: Cannot read property 'length' of undefined

```
At C:\Users\frocha\.atom\packages\pymakr\node_modules\xterm\lib\utils\BufferLine.js:11

TypeError: Cannot read property 'length' of undefined
    at Object.translateBufferLineToString (/packages/pymakr/node_modules/xterm/lib/utils/BufferLine.js:11:29)
    at /packages/pymakr/node_modules/xterm/lib/SelectionManager.js:109:45)
    at Terminal.getSelection (/packages/pymakr/node_modules/xterm/lib/xterm.js:883:57)
    at Term.copy (/packages/pymakr/lib/main/terminal.js:162:34)
    at Term.termKeyPress (/packages/pymakr/lib/main/terminal.js:103:16)
    at /packages/pymakr/lib/main/terminal.js:32:15)
    at Terminal.EventEmitter.emit (/packages/pymakr/node_modules/xterm/lib/EventEmitter.js:49:20)
    at Terminal.keyDown (/packages/pymakr/node_modules/xterm/lib/xterm.js:928:10)
    at /packages/pymakr/node_modules/xterm/lib/xterm.js:365:14)
```

### Commands

```
     -7:58.2.0 core:copy (textarea.xterm-helper-textarea)
```

### Non-Core Packages

```
pymakr 1.2.7 
```

RalphHogenbirk, 2018-01-16:
I've seen this once before, happened randomly but isn't reproducible in my case. Does it keep happening in your case, or was it just once? Anyway it seems to be an issues inside the xterm lib, triggered when the pymakr code asks the xterm object for the selection. Will look into this when I get the chance.

RalphHogenbirk, 2018-02-28:
It's a bug in xterm v2 and it's fixed in xterm v3. I'm working to intergrate it into pymakr, which takes some time. I have to re-write some of the code because of big changes in xterm v3. Will close this thread when it's fixed and released.

saisie-mobile, 2018-08-08:
append today at connection and disconnection and reconnection of SiPy at power on of the card; after : no way.

jirikrepl, 2020-02-03:
appeared today after right clickicking on a large amount of text or pressing cmd+c

[image]

elmasria, 2021-01-15:
A new version exists that solves the above issue. if you face it again, please reopen the issue.

closed 2021-01-15 by elmasria