pycom/pymakr-atom #64: Uncaught TypeError: selected_tree.getPath is not a function

StridesIT, 2018-01-29:
[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.23.3 x64
**Electron**: 1.6.15
**OS**: Mac OS X 10.12.6
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.2.7

### Stack Trace

Uncaught TypeError: selected_tree.getPath is not a function

```
At /Users/rameshmurthy/.atom/packages/pymakr/lib/main/api-wrapper.js:69

TypeError: selected_tree.getPath is not a function
    at ApiWrapper.getProjectPath (/packages/pymakr/lib/main/api-wrapper.js:69:32)
    at /packages/pymakr/lib/main/api-wrapper.js:8:30)
    at /packages/pymakr/lib/board/sync.js:15:16)
    at Pymakr.sync (/packages/pymakr/lib/pymakr.js:309:22)
    at /packages/pymakr/lib/pymakr.js:58:15)
    at emitNone (events.js:86:13)
    at PanelView.emit (events.js:185:7)
    at HTMLButtonElement.PanelView.button_sync.onclick (/packages/pymakr/lib/main/panel-view.js:107:13)
```

### Commands

```
     -7:35.2.0 application:add-project-folder (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
  2x -1:36.9.0 core:confirm (atom-workspace.workspace.scrollbars-visible-when-scrolling.theme-one-dark-syntax.theme-one-dark-ui)
     -0:58.9.0 core:copy (atom-notification.fatal.icon.icon-bug.native-key-bindings.has-detail.has-close.has-stack)
```

### Non-Core Packages

```
atom-ide-ui 0.7.2 
busy-signal 1.4.3 
gcc-make-run 0.2.12 
intentions 1.1.5 
language-elm 1.5.0 
linter 2.2.0 
linter-elm-make 0.28.1 
linter-ui-default 1.6.10 
nuclide 0.278.0 
pymakr 1.2.7 
```

RalphHogenbirk, 2018-02-05:
Hi, thanks for reporting. I'll at least put an extra check in the code to prevent this from going wrong. 

Do you have any info I can use to reproduce this exact issue? Did you disable the tree-view on the left, or change any setting concerning the tree-view?

RalphHogenbirk, 2018-02-28:
Should be fixed in the latest release (1.2.8). If not, feel free to re-open the issue.

closed 2018-02-28 by RalphHogenbirk