pycom/pymakr-atom #65: Error: Writing to COM port: Upload failed. Please reboot your device manually

SpoofEx, 2018-01-31:
It seems I have a similar issue as discussed in [with](https://github.com/pycom/pymakr-atom/issues/45) uploading the simplest project with only a main.py file containing print("Hello World"):
[image]
Current configuration:
windows 10 1709 16299.125
atom 1.23.3
Pymakr 1.2.27

I moved back to Pymaker 1.1.0 as Smeedy suggested, but the communication via COM6 with the ESP directly hangs:
[image]
clicking on 'more' > 'Get serial ports' results in:
[image]

However using ampy works fine to get things transported to the board and vice versa, but it would be much nicer if pymkr would work so that my workflow gets integrated.

bjoernbusch, 2018-02-01:
I have a similar issue on MacOS where I can't upload changes.

[image]

I have to reset the flash and do a full load, which is quite annoying.

TamiasSibiricus, 2018-02-03:
Same thing on ubuntu 16.04  with vscode plugin.

RalphHogenbirk, 2018-02-05:
Hi @SpoofEx are you on the latest firmware? There have been a bunch of stability improvements between the firmware and pymakr in the last versions. Going back to 1.1.0 indeed also doesn't work for all firmware versions. Let me know if updating helps.

@bjoernbusch this happens on older boards that have less RAM. The current way of uploading takes a bit of memory, so if you are already using some memory it will run out. It will be solved in the next upcoming Atom + firmware release, where we'll put the board in safe boot automatically for each sync, if there is not enough memory available.

SpoofEx, 2018-02-05:
@RalphHogenbirk isn't pymakr 1.2.7  the latest version? At least that is the one I find as the most actual in my atom installation if not how do I find the latest version?

bjoernbusch, 2018-02-05:
Thanks for the clarification. How can I tell whether the board is older? Version, SerialNumber, etc?

RalphHogenbirk, 2018-02-06:
@SpoofEx I'm talking about the firmware on the board itself. You can see the version using 'more -> get firmware version'. Update it using the firmware updater tool from https://pycom.io/downloads/. Firmware updates are coming rapidly and sometimes with important fixes, so it's a good thing to keep up to date :)

@bjoernbusch It's complicated, since all boards got an upgrade at some point. But you can see how much ram is avialable with os.getfree('/flash') (it returns Kb's). Try it after a safe boot or when you have no code running. If you get close to 4000Kb, it's a new board. Old boards should have about 96Kb available.

SpoofEx, 2018-02-07:
@RalphHogenbirk I am using this  [ESP32 Board](https://www.aliexpress.com/item/ESP-32-ESP-32S-WiFi-Bluetooth-Development-Board-2-4GHz-Micro-USB-CP2102-Version-Dual-Core/32799845452.html) and flashed the firmware with the latest micropython [firmware](http://micropython.org/resources/firmware/esp32-20180207-v1.9.3-278-gcc92c057.bin) using the esptool.

RalphHogenbirk, 2018-02-07:
@SpoofEx Aha, so not a Pycom board. The code we are using for the upload feature is partly dependent on the Pycom specific firmware, and will probably not work on just bare micropython. We're planning to rebuild the upload feature in Pymakr, after which it will likely also work on your board, but until then you'll have to revert to other upload methods like FTP or [rshell](https://github.com/dhylands/rshell).

CodeManiac64, 2018-02-21:
I'm currently using Windows 10, and I can upload just fine to my PyTrack. But when I want to switch to PySense I encounter the same error as @SpoofEx.
Anyone know why this is the case for only PySense and not for PyTrack? Could it be the pymakr 1.2.7 that cause the error? Everything is updated and am using the lastest versions.
I've been able to upload before to both devices, this error happend just recently.

RalphHogenbirk, 2018-02-28:
Guys, we've just released pymakr 1.2.8 that might be of use to some of you. First of all, it has a few bugfixes concerning the 'upload failed' error. For people having memory allocation issues when uploading code, it has a new option in the settings called 'safe boot before upload', which prevents this error. Only works with the latest firmware though, so be sure to upgrade.

Let me know if it helps any of you.

bjoernbusch, 2018-03-05:
I just updated to the latest plugin and the latest FW (17), but when I enable the option in the global settings in only get: Safe boot not available for this version ()

Any hints?

CodeManiac64, 2018-03-05:
@bjoernbusch 
I had the same issue with uploading my code to my LoPy device. What I needed to do was manually doing a safe boot in order to be able to upload.
The annoying thing is that if you decide to switch device, you'd have to do the manuall safe boot again...
Have not yet found out why that's the case for me, nor how to fix it completely

RalphHogenbirk, 2018-03-07:
Looking at the output you get @bjoernbusch, this must be a bug in Pymakr. Are you both on windows and using a LoPy1?

bjoernbusch, 2018-03-07:
No, I'm on a Mac, but yes, it's a LoPy1.

RalphHogenbirk, 2018-03-10:
Thanks. My suspicion is that this happens when asking for the version number already throws a memory exception. In any case, the error handling is not good at that point in the code. I'll come back to you on this after the weekend.

bjoernbusch, 2018-03-19:
ok, so I updated to the latest version of everything and I still have the same issue. I could narrow it down to a file which is apparently causing the issue. In that file, I have a class defined, without any imports. When I add an import (I tried both pycom and struct), the file is uploaded properly. Is that a known issue / limitation?

RalphHogenbirk, 2018-03-20:
@bjoernbusch still the "Safe boot not available for this version ()"? What error message does it throw when that one file fails to upload?

sdaniel55, 2018-07-09:
Hi @bjoernbusch 👋 , could you update the issue? Are you still having problems? Note: the latest version of Atom is `1.2.8` and pymakr `1.4.0`

njordan77, 2018-11-03:
having the same issue. ATOM+pymkr suck or i have no idea what i'm doing. COM connects but timeout when trying to upload a simple file. (LoPy Lora). after flashing with pycom firmware all is fine again for the first time....after that no upload at all.....

solving this issue would be highly appreciated! Thanks Norbert

RalphHogenbirk, 2018-11-07:
@njordan77 Very likely a different issue than what's going on above. Can you create a new issue with some more information on which versions of pymakr, Atom and firmware you are using. A screenshot of the terminal this error happens would also help. We'll try to help you solve the issue asap!

pirate, 2018-11-10:
I'm having this issue as well in VSCode.  I don't think it's Atom-specific.

[image]

It seems whatever program is uploaded first stays on the board and can't be interrupted for subsequent uploads.

RalphHogenbirk, 2018-11-12:
Hi @pirate, in your case it looks like the safe boot doesn't work properly. Can you double check that by pressing ctrl-f before using the upload button? Ctrl-f should safe boot the device and the gps script you seem to be running should not start. If that doesn't happen, let me know what firmware version and hardware you are using. Worst case you might need to safe boot manually (connecting P12 to 3.3v and rebooting)

pirate, 2018-11-13:
Thanks, I just managed to get it working by pressing the reset button for 12s, apologies if this was unrelated to the main issue.

prashanthmeda, 2019-06-13:
@SpoofEx if you get a REPL prompt in Putty and can enter commands,. You have to disable flow control in the Putty serial interface settings. Try something like Ctrl-D, which should reboot the device. Then tyr Ctrl-B should show the firmware revision. Once >>> you get this option try entering any Python statement, like:
print("Hello World")
It worked for me

closed 2020-02-18 by pedroknup