pycom/pymakr-atom #69: Failed to install Pymakr

liowmei, 2018-03-16:
Hi there, i am using Windows 8.1. My atom version is 1.25.0 while my pymakr plugin that i want to install stated there 1.2.9 but when i click it it give me 1.0.3 which stated compatible to my Atom version. But in the end i failed to install the Pymakr. How can i solve this problem?

sdaniel55, 2018-07-09:
Hi @liowmei, did you manage to install Pymakr? I'll close this issue, but in case you didn't, feel free to reopen it and I'll help you out. Thanks!

closed 2018-07-09 by sdaniel55

cirobr, 2019-08-13:
Hello,
Running Win10-64 and Atom 1.40.0 x64. Pymakr 1.4.14 is not enabled for installation although the error message shows my Atom version is OK for it. Thanks for advising.