pycom/pymakr-atom #70: Impossible to install Pymakr on ATOM

SpaekEasy, 2018-03-16:
Hello, 

I use Pymakr package since month with Atom and since this morning it's doesn't work anymore.

First i had in a red box a miskate saying that Atom was not able to launch Pymark. So i uninstall it and now i can't install it anymore.

I have this :+1: 
[image]

RalphHogenbirk, 2018-03-16:
We're working on getting Pymakr compatible with atom 1.25 right now, sorry for the inconvenience! Expect an update later today.

It says pymakr 1.0.3 as latest compatible because it's the last Pymakr version that didn't have any version compatibility info yet.

SpaekEasy, 2018-03-16:
Oh ok !

So i need to get an older version of Atom to make Pymark package working with Atom ?

RalphHogenbirk, 2018-03-16:
Yes you can download 1.24.0 from https://github.com/atom/atom/releases and it should work right now.

SpaekEasy, 2018-03-16:
Unfortunately this is not working... it's still can't install Pymakr. I have this message now :

RalphHogenbirk, 2018-03-16:
@SpaekEasy I've released v1.2.10 right now which should work with atom 1.25 on osx and windows 64. Can you try it and let me know if it fixes your issues?

closed 2018-03-17 by RalphHogenbirk

giampiero7, 2018-05-22:
Hi,
Same problem now with version 1.27.1 of Atom

RalphHogenbirk, 2018-05-22:
Got it @giampiero7, will release a fix soon with support for 1.27.1. You could downgrade to 1.27.0 for now. i'll update you when the release is out

RalphHogenbirk, 2018-05-23:
Pymakr v1.3.2 should now install properly on atom v1.27.1

achalkasturia, 2020-10-21:
Same Issue with atom 1.52.0x64