pycom/pymakr-atom #71: pymakr-atom

Manzi12, 2018-03-17:
Installing “pymakr@1.2.10” failed.Show output…

RalphHogenbirk, 2018-03-17:
Hi @Manzi12 can you copy a print of the output? Which OS are you on? Windows32 and linux have an issue with Atom 1.25 right now, which is solvable by downgrading to 1.24. Expect a fix on monday

Manzi12, 2018-03-17:
"Installing “pymakr@1.2.10” failed.Show output... "That's all it says and interestingly i'm not using any of the OS that you mentioned.i'm using Windows 64 bit

RalphHogenbirk, 2018-03-17:
Normally you can click the 'show output' part and it will show you some more info about the error that occurred. That's the info I need to help debug the issue.

Manzi12, 2018-03-18:
Installing “pymakr@1.2.10” failed.Hide output…

npm ERR! Windows_NT 10.0.16299
npm ERR! argv "C:\\Users\\manziwit\\AppData\\Local\\atom\\app-1.24.1\\resources\\app\\apm\\bin\\node.exe" "C:\\Users\\manziwit\\AppData\\Local\\atom\\app-1.24.1\\resources\\app\\apm\\node_modules\\npm\\bin\\npm-cli.js" "--globalconfig" "C:\\Users\\manziwit\\.atom\\.apm\\.apmrc" "--userconfig" "C:\\Users\\manziwit\\.atom\\.apmrc" "install" "C:\\Users\\manziwit\\AppData\\Local\\Temp\\d-118218-12500-1n99p26.mcy5ezh0k9\\package.tgz" "--runtime=electron" "--target=1.6.16" "--arch=x64" "--global-style"
npm ERR! node v6.9.5
npm ERR! npm  v3.10.10
npm ERR! file C:\Users\manziwit\.atom\.apm\rimraf\2.6.2\package\package.json
npm ERR! code EJSONPARSE

npm ERR! Failed to parse json
npm ERR! Unexpected token '\u0000' at 1:1
npm ERR! 
npm ERR! ^
npm ERR! File: C:\Users\manziwit\.atom\.apm\rimraf\2.6.2\package\package.json
npm ERR! Failed to parse package.json data.
npm ERR! package.json must be actual JSON, not just JavaScript.
npm ERR! 
npm ERR! This is not a bug in npm.
npm ERR! Tell the package author to fix their package.json file. JSON.parse

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\manziwit

there you have it.

barsznica, 2018-04-13:
> EPERM: operation not permitted, unlink 
 C:\Users\peter.barsznica\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node' 

I hate PyCom!

RalphHogenbirk, 2018-04-30:
Hey @Manzi12, did you manage to solve this issue? I don't yet understand why it would throw an error on the rimraf package, which is installed inside apm itself. Try installing the new version of Pymakr (V1.3.0) on the latest version of Atom and see if that fixes it. If it's still throwing this error, check if you can indeed spot something weird in the package.json file this is thrown from (```C:\Users\manziwit.atom.apm\rimraf\2.6.2\package\package.json```)

sdaniel55, 2018-07-09:
Hi @Manzi12 👋 
Could you update this issue? Did you manage to solve it? Thanks!

Manzi12, 2018-07-11:
Hi

I didn't manage to solve it so instead i used a different hardware

RalphHogenbirk, 2018-07-11:
@Manzi12 That's too bad, but why would you use different hardware? You can use the board fine even without Pymakr, using FTP or mpfshell for uploading, and putty or other terminal program for using REPL. There's also Pymakr for VSCode as an alternative.

Let me know if you want any help solving this issue anyway.

closed 2020-02-18 by pedroknup