pycom/pymakr-atom #73: Failed to load the pymakr package

robmarkcole, 2018-03-19:
On following instructions to update pymakr on Mac High Sierra 10.13.3 I get the following:

[image]

The referenced issue is https://github.com/pycom/pymakr-atom/issues/53

```
The module '/Users/robincole/.atom/packages/Pymakr/node_modules/bindings/build/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
Hide Stack Trace
Error: The module '/Users/robincole/.atom/packages/Pymakr/node_modules/bindings/build/serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at bindings (/Users/robincole/.atom/packages/Pymakr/node_modules/bindings/bindings.js:76:44)
    at Object.<anonymous> (/Users/robincole/.atom/packages/Pymakr/node_modules/serialport/lib/bindings.js:3:35)
    at Object.<anonymous> (/Users/robincole/.atom/packages/Pymakr/node_modules/serialport/lib/bindings.js:36:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/robincole/.atom/packages/Pymakr/node_modules/serialport/lib/serialport.js:12:25)
    at Object.<anonymous> (/Users/robincole/.atom/packages/Pymakr/node_modules/serialport/lib/serialport.js:476:3)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (/Applications/Atom.app/Contents/Resources/app/src/compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///Applications/Atom.app/Contents/Resources/app.asar/static/index.js:47:45)
    at require (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:66:33)
    at Object.<anonymous> (/Users/robincole/.atom/packages/pymakr/lib/connections/pyserial.js:21:18)
    at Object.<anonymous> (/Users/robincole/.atom/packages/pymakr/lib/connections/pyserial.js:27:30)
    at Module.get_Module._compile (/Applications/Atom.app/Contents/Resources/app/src/native-compile-cache.js:106:36)
```

RalphHogenbirk, 2018-03-19:
Looks like you're using atom 1.24? Should work on Atom 1.25, for which the serialport library has been precompiled.

robmarkcole, 2018-03-19:
@RalphHogenbirk updating to 1.25 fixed the issue, thanks!

closed 2018-03-19 by robmarkcole