pycom/pymakr-atom #74: Failed to activate the pymakr package

Rachelsimida, 2018-03-28:
Failed to activate the pymakr package
The module '\\?\C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
Hide Stack Trace
Error: The module '\\?\C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\build\Release\serialport.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 53. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Rachel/AppData/Local/atom/app-1.24.0/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\native-compile-cache.js:66:33)
    at bindings (C:\Users\Rachel\.atom\packages\pymakr\node_modules\bindings\bindings.js:76:44)
    at Object.<anonymous> (C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings.js:3:35)
    at Object.<anonymous> (C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\lib\bindings.js:36:3)
    at Module.get_Module._compile (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Rachel/AppData/Local/atom/app-1.24.0/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\lib\serialport.js:12:25)
    at Object.<anonymous> (C:\Users\Rachel\.atom\packages\pymakr\precompiles\serialport-win\lib\serialport.js:476:3)
    at Module.get_Module._compile (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\native-compile-cache.js:106:36)
    at Object.value [as .js] (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\compile-cache.js:239:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (file:///C:/Users/Rachel/AppData/Local/atom/app-1.24.0/resources/app.asar/static/index.js:47:45)
    at require (C:\Users\Rachel\AppData\Local\atom\app-1.24.0\resources\app\src\native-compile-cache.js:66:33)
    at Object.<anonymous> (file:///C:/Users/Rachel/.atom/packages/pymakr/lib/connections/pyserial.js:21:18)
    at Object.<anonymous> (file:///C:/Users/Rachel/.atom/packages/pymakr/lib/connections/pyserial.js:27:30)
    at Module.get_Module._compile (C:\Users\Rachel\AppData\Loc

Rachelsimida, 2018-03-28:
Pymakr vision: 1.2.11

RalphHogenbirk, 2018-04-17:
Hi @Rachelsimida, very late response on this, but it might be that you are on an old version of Atom. Please try to update to the newer versions (1.25.0 or 1.25.1). Let me know if that helps

steek1337, 2018-04-25:
apm install pymakr@1.2.12 I did get that with atom 1.24.1 x64 and still not working

RalphHogenbirk, 2018-04-25:
@steek1337 I was working on the windows issue today and have found some issues with the install script. Our plan is to release a new version Friday that includes a fix for it. I'll let you know when it's out.

steek1337, 2018-05-03:
[image]

I dont se the terminal... do I have to add then in another way now!?
@RalphHogenbirk

RalphHogenbirk, 2018-05-04:
You should definitely see the terminal. You got this after updating to v1.3.0? Can you check the console (ctrl+alt+i) if you see any errors? Otherwise try uninstalling and re-installing again.

HenroRitchie, 2018-05-10:
Hi 

I installed v1.3.0 and atom opened without generating any errors. I then connected/updated a wipy 3.0 module and now atom generates the following error when opening it. I reinstalled it a couple of times without any success.

[Enter steps to reproduce:]

1. ...
2. ...

**Atom**: 1.26.1 x64
**Electron**: 1.7.11
**OS**: Microsoft Windows 10 Pro
**Thrown From**: [pymakr](https://github.com/pycom/pymakr-atom) package 1.3.0

### Stack Trace

Failed to activate the pymakr package

```
At Could not locate the bindings file. Tried:
 → C:\Users\HenroRitchie\node_modules\bindings\build\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\out\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\default\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\compiled\7.9.0\win32\x64\serialport.node

Error: Could not locate the bindings file. Tried:
 → C:\Users\HenroRitchie\node_modules\bindings\build\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\out\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\Debug\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\out\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\Release\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\build\default\serialport.node
 → C:\Users\HenroRitchie\node_modules\bindings\compiled\7.9.0\win32\x64\serialport.node
    at bindings (~/node_modules/bindings/bindings.js:93:9)
    at /packages/Pymakr/precompiles/serialport-win/lib/bindings.js:3:35)
    at /packages/Pymakr/precompiles/serialport-win/lib/bindings.js:36:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/Pymakr/precompiles/serialport-win/lib/serialport.js:12:25)
    at /packages/Pymakr/precompiles/serialport-win/lib/serialport.js:476:3)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/connections/pyserial.js:21:18)
    at /packages/pymakr/lib/connections/pyserial.js:27:30)
    at Module.get_Module._compile (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:106:36)
    at Object.value [as .js] (~/AppData/Local/atom/app-1.26.1/resources/app/src/compile-cache.js:240:29)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (/app.asar/static/index.js:47:45)
    at require (~/AppData/Local/atom/app-1.26.1/resources/app/src/native-compile-cache.js:66:33)
    at /packages/pymakr/lib/board/pyboard.js:3:34)
    at /packages/pymakr/lib/board/pyboard.js:28:29)
```

### Commands

```
     -2:15.7.0 pymakr:toggleREPL (input.hidden-input)
     -1:49.6.0 core:confirm (input.hidden-input)
```

### Non-Core Packages

```
pymakr 1.3.0 
```