pycom/pymakr-atom #77: Uncaught TypeError: No Binding set on SerialPort.Binding

tusharppatel, 2018-04-12:
I  can't make connection to FiPY and getting error "Uncaught TypeError: No Binding set on SerialPort.Binding" on Atom when click on "Get Serial Port"

pymakr:1.2.12
Atom:1.25.1 ia32
FiPY firmware: 1.17.3.b1(checked on other PC)

error detail:
Uncaught TypeError: No Binding set on SerialPort.Binding
C:\Users\tushar.patel\.atom\packages\pymakr\precompiles\serialport-win32\lib\serialport.js:614
Hide Stack Trace
TypeError: No Binding set on `SerialPort.Binding`
    at Function.SerialPort.list (C:\Users\tushar.patel\.atom\packages\pymakr\precompiles\serialport-win32\lib\serialport.js:614:11)
    at Function.list (file:///C:/Users/tushar.patel/.atom/packages/pymakr/lib/connections/pyserial.js:143:16)
    at Pymakr.getSerial (file:///C:/Users/tushar.patel/.atom/packages/pymakr/lib/pymakr.js:176:14)
    at PanelView.<anonymous> (file:///C:/Users/tushar.patel/.atom/packages/pymakr/lib/pymakr.js:87:13)
    at emitNone (events.js:86:13)
    at PanelView.emit (events.js:188:7)
    at HTMLDivElement.PanelView.option_get_serial.onclick (file:///C:/Users/tushar.patel/.atom/packages/pymakr/lib/main/panel-view.js:153:13)

RalphHogenbirk, 2018-04-13:
Hey @tusharppatel, thanks for reporting. Looks like a problem with the serialport library on windows32. Let me check this on my windows machine and get back to you when I find the issue. Sorry for the inconvenience.

RalphHogenbirk, 2018-04-30:
@tusharppatel Pymakr V1.3.0 is out now, that should fix some of these issues. Let me know if it solves yours.

closed 2020-02-18 by pedroknup